JdeRobot/RoboticsAcademy #807: grammatical error on the front page of the website

dgvsnl, 2021-04-03:
[image]

> there is an grammatical error in second line of second paragraph.

> Here's the link if anyone wants refer -- [link](http://jderobot.github.io/RoboticsAcademy/)

dgvsnl, 2021-04-03:
Here is my suggestion:

Instead of --
"The student inserts her code in the template file...."

We can write --
**"In which student can insert their code in the templete file...."**

dgvsnl, 2021-04-04:
Can you guide me **@jmplaza** where I can make this kind of change in this repository

trunc8, 2021-04-04:
@dgvsnl you can follow these steps:
1. Fork the RoboticsAcademy repository
2. Clone the forked copy
3. Your changes would be in this file location- [docs](https://github.com/JdeRobot/RoboticsAcademy/blob/master/docs/_pages/home.md)
4. Commit and push the changes to your own copy
5. Submit a pull request to this repo

dgvsnl, 2021-04-04:
Thanks for the help @trunc8

jmplaza, 2021-04-04:
the issue should not be closed once the Pull-Request (PR) is sent. Only when the issue is really solved (for instance when the PR is merged, etc).