JdeRobot/RoboticsAcademy #813: Refactor web templates to share common parts

jmplaza, 2021-04-04:
This way, the exercise.html of many exercises will share common widgets, assets and pieces of JS code.

jmplaza, 2021-04-04:
For instance ws_code.js file

jmplaza, 2021-04-12:
The long term solution for this will come with the refactoring of all exercise frontend from vanillaJS to ReactJS.

Meanwhile exploring the symlinks may be a good fix. This way all the exercise will have a "base.html" on their directories, but they all will point to a base.html in "RoboticsAcademy/exercise/assets"... It should work for Linux, Windows and MacOS operating systems. That is a requirement. What do you think about this possibility @SakshayMahna @dvalladaresv ?

SakshayMahna, 2021-04-12:
But, would the symlinks that are generated for a Linux machine work on a Windows machine or vice versa. From what I read, the symlink generation for both seems different, [this link](https://superuser.com/questions/528753/convert-linux-symlinks-to-windows-symlinks). 

Not sure, but can we have a look at .tpl files? Here is a [link](https://stackoverflow.com/questions/1808294/what-are-tpl-files-php-web-design)

PR #917 (2021-05-07): Refactoring of templates with common parts
Fix #813
closed 2021-05-07 by ReyDoran