JdeRobot/RoboticsAcademy #814: FollowLine exercise for ROS1-Noetic

jmplaza, 2021-04-04:
As an exploratory exercise in the latest (and last...) release of ROS1. It involves:
- [ ] creation of a new RADI, named Noetic-RADI
- [ ] refactoring of exercise.py to python3
- [ ] adaptation of manager (maybe a new manager-noetic) to launch Noetic-based exercises 
while keeping the same exercise.html, and the same vnc-based Gazebo 3D viewer

Suhas-G, 2021-04-05:
I'll start working on it. If I have any questions or need any help I'll open a thread on the forum. Thanks!

trunc8, 2021-04-09:
Hi @jmplaza and @Suhas-G, I would like some guidance.
[image]
I could establish the connection between the web template and the RADI. I have tried to make minimal changes to the existing Melodic-RADI. I have written a verbose [commit message](https://github.com/trunc8/RoboticsAcademy/commit/11878d7ee2fa53de88dabc08fd094e35702b8250) describing my changes.
Steps to replicate:
- `git clone https://github.com/trunc8/RoboticsAcademy.git && cd RoboticsAcademy && git checkout noetic-RADI`
- `cd scripts && docker build . -t noetic_jde_image`
- `docker run -it
      --rm
      --name=noetic_jde_container
      -p 8080:8080 -p 7681:7681 -p 2303:2303 -p 1905:1905 -p 8765:8765 -p 6080:6080
      noetic_jde_image python3.8 manager-noetic.py`

Problem: Within a few seconds of entering my solution and pressing "Play", the Brain frequency measure goes to 0.
[image]

These are some snippets of errors from the terminal which may be relevant:
[image]
[image]

Suhas-G, 2021-04-09:
Hello,
So there are 2 issues.
a. NoVNC is not getting started. This is because the websockify latest release (v0.9) calls 'python -m websockify' instead of python3. The master branch of websockify already has this fix. So we can either use master branch or use a pckage like 'python-is-python3'. That should make it work
b. The gazebo errors. I'm also getting the same errors. Even though the errors are present gazebo started fine but the angle of the car is wrong.
[image]

From the error messages its supposed to be related to images captured from camera and services being initialised more than once, so I'm not sure whether the rotation of car is because of this. So this issue I'm not sure and still trying to fix the issue

trunc8, 2021-04-09:
Right! I can view Gazebo with the master branch fix for websockify.
[image]  
There are two more errors in the above screenshot in addition to the perpendicular F1 car:
- The red blip in the Track map is nowhere close to the actual position
- The console window shows a '127.0.0.1 refused to connect.' message

@Suhas-G, could these errors be related to the terminal error messages?

Suhas-G, 2021-04-09:
@trunc8:
The errors on terminal about topics and the red blip being wrong are both because the 'f1' model in [melodic-devel](https://github.com/JdeRobot/CustomRobots/blob/melodic-devel/f1/f1/model.sdf) branch and [noetic-devel](https://github.com/JdeRobot/CustomRobots/blob/noetic-devel/f1/models/f1/model.sdf) branch are quite different. Maybe it has to be fixed in CustomRobots repo... @jmplaza Can you please advise on this?
Another thing is even after copying the model.sdf file from melodic-devel I had to give an extra yaw of 90 degrees like so
```
<link name="f1">
      <pose>0 0 0 0 0 1.57</pose>
....
```
So I'm not completely sure about what are all the differences in 2 branches.

The console issue I think is probably because you didn't specify port mapping of 1108 in 
`docker run -it --rm --name=noetic_jde_container -p 8080:8080 -p 7681:7681 -p 2303:2303 -p 1905:1905 -p 8765:8765 -p 6080:6080 noetic_jde_image python3.8 manager-noetic.py`

trunc8, 2021-04-09:
Yes! @Suhas-G I figured out the port fix too just some time back! Thanks

> From the error messages its supposed to be related to images captured from camera and services being initialised more than once, so I'm not sure whether the rotation of car is because of this. So this issue I'm not sure and still trying to fix the issue

Based on [this link](https://answers.ros.org/question/12244/complaint-from-openni_launch-on-already-advertised-services/?answer=18074#post-id-18074) it seems the errors are of low significance and can be ignored for the time being

trunc8, 2021-04-09:
I finally finished simulating a lap in the Noetic-RADI! In the Dockerfile, I switched the CustomRobots branch back to `melodic-devel` (to make it work for the time being). All my modifications are visible in the `noetic-RADI` branch of my [forked repo](https://github.com/trunc8/RoboticsAcademy/commits/noetic-RADI) dated Apr 9, 2021.

@Suhas-G, you have been working on this issue longer. It would be best if you made the PR for this :)

@jmplaza, shall we maintain the Noetic Dockerfile in the master branch or a separate branch?

jmplaza, 2021-04-09:
We better create a noetic-branch (the current master is for ROS Melodic) as the Dockerfile will be different and also the exercise.py of the exercises. In Noetic we will use Python 3.X instead of the default Python 2.7 of melodic-branch

Suhas-G, 2021-04-10:
@jmplaza  Can you create a noetic branch on the repo? I will raise the pull request to that itself.

jmplaza, 2021-04-10:
Done. It only has the FollowLine exercise. Feel free to update the exercise.py to Python3 and the Dockerfile to create a RADI working with it

jmplaza, 2021-04-12:
Hi @Suhas-G,
there is no hurry to migrate FollowLine exercise from ROS-Melodic to ROS-Noetic, so let's focus on improving also its dependencies on CustomRobots repository, to be pure ROS-Noetic too. This way we will avoid further refactoring and keep the Melodic and the Noetic side fully separated. Feel free to create any required issue and PR on CustomRobots repository regarding its noetic-branch

Suhas-G, 2021-04-12:
ok @jmplaza 
I compared few files from both melodic and noetic branches of CustomRobots, but I don't know why the changes were introduced. I'll try to go through commit history and see why it has diverged.

Suhas-G, 2021-04-16:
@jmplaza  The PR on CustomRobots got merged. So I have updated this issue [PR](https://github.com/JdeRobot/RoboticsAcademy/pull/846) to have Dockerfile which pulls the`noetic-devel` branch.

jmplaza, 2021-04-29:
Using RADI-3.0.0 #904

pariaspe, 2021-06-15:
Solved in #1027

closed 2021-06-15 by pariaspe