JdeRobot/RoboticsAcademy #833: [RADI] Performance Issue and bugs

Jovian-Dsouza, 2021-04-07:
Exercise : Drone Cat and Mouse
My System : Ryzen 5 5500u + gtx 1650 laptop
I am using the docker image built from the latest Dockerfile in this repo

The FPS of the simulation is very low (around 4.8 FPS). Doing an `nvidia-smi` doesn't show gpu utilization from the docker container.

Also the simulation stops suddenly presenting with this error on the docker output.

```
['/', '/opt/ros/melodic/lib/python2.7/dist-packages', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/usr/local/lib/python3.8/dist-packages', '/usr/lib/python3/dist-packages']
<websockets.server.WebSocketServerProtocol object at 0x7f72a15aaf70>
{"command":"exit","exercise":""}
ALL KILED
{"command":"open","exercise":"drone_cat_mouse","width":"890","height":"459"}
> Starting simulation
_XSERVTransmkdir: ERROR: Cannot create /tmp/.X11-unix
_XSERVTransSocketUNIXCreateListener: mkdir(/tmp/.X11-unix) failed, errno = 17
_XSERVTransMakeAllCOTSServerListeners: failed to create listener for unix

X.Org X Server 1.19.6
X.Org X Server 1.19.6
Release Date: 2017-12-20
X Protocol Version 11, Revision 0

Release Date: 2017-12-20
Build Operating System: Linux 4.15.0-124-generic x86_64 Ubuntu
X Protocol Version 11, Revision 0
Current Operating System: Linux 6e741f5d0c0c 5.11.11-051111-generic #202103310025 SMP Thu Apr 1 01:54:13 UTC 2021 x86_64
Build Operating System: Linux 4.15.0-124-generic x86_64 Ubuntu
Current Operating System: Linux 6e741f5d0c0c 5.11.11-051111-generic #202103310025 SMP Thu Apr 1 01:54:13 UTC 2021 x86_64
Kernel command line: Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.11.11-051111-generic root=UUID=8e0f3578-bc45-487c-b39BOOT_IMAGE=/boot/vmlinuz-5.11.11-051111-generic root=UUID=8e0f3578-bc45-487c-b391-55b29bb28286 ro quiet splash acpi_osi=Linux acpi_backlight=video vt.handoff=7
1-55b29bb28286 ro quiet splash acpi_osi=Linux acpi_backlight=video vt.handoff=7
Build Date: 30 November 2020  08:01:56PM
Build Date: 30 November 2020  08:01:56PM
xorg-server 2:1.19.6-1ubuntu4.8 (For technical support please see http://www.ubuntu.com/support) 
xorg-server 2:1.19.6-1ubuntu4.8 (For technical support please see http://www.ubuntu.com/support) 
Current version of pixman: 0.34.0
Current version of pixman: 0.34.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: Markers: (--) probed, (--) probed, (**) from config file, (**) from config file, (==) default setting,
	(==) default setting,
	(++) from command line, (++) from command line, (!!) notice, (!!) notice, (II) informational,
	(II) informational,
	(WW) warning, (WW) warning, (EE) error, (EE) error, (NI) not implemented, (NI) not implemented, (??) unknown.
(??) unknown.
(++) Log file: "./console_xdummy.log", Time: Wed Apr  7 10:25:09 2021
(++) Log file: "./xdummy.log", Time: Wed Apr  7 10:25:09 2021
(++) Using config file: "./xorg.conf"
(++) Using config file: "./xorg.conf"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
Gazebo multi-robot simulator, version 9.16.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Wrn] [gazebo_gps_plugin.cpp:76] [gazebo_gps_plugin]: iris::gps0 using gps topic "gps0"
[Wrn] [gazebo_gps_plugin.cpp:201] [gazebo_gps_plugin] Using default update rate of 5hz 
[Wrn] [gazebo_gps_plugin.cpp:76] [gazebo_gps_plugin]: iris_red::gps0 using gps topic "gps0"
[Wrn] [gazebo_gps_plugin.cpp:201] [gazebo_gps_plugin] Using default update rate of 5hz 
07/04/2021 10:25:14 07/04/2021 10:25:14 passing arg to libvncserver: -rfbport
passing arg to libvncserver: -rfbport
07/04/2021 10:25:14 07/04/2021 10:25:14 passing arg to libvncserver: 5900
passing arg to libvncserver: 5901
07/04/2021 10:25:14 07/04/2021 10:25:14 x11vnc version: 0.9.13 lastmod: 2011-08-10  pid: 903
x11vnc version: 0.9.13 lastmod: 2011-08-10  pid: 914
07/04/2021 10:25:14 Using X display :0
[... 178 lines cut ...]
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/opt/jderobot/CustomRobots/3d_reconstruction/launch"
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/opt/jderobot/CustomRobots/3d_reconstruction/words"
172.17.0.1 - - [07/Apr/2021 10:25:26] 172.17.0.1: Plain non-SSL (ws://) WebSocket connection
172.17.0.1 - - [07/Apr/2021 10:25:26] 172.17.0.1: Path: '/websockify'
172.17.0.1 - - [07/Apr/2021 10:25:26] connecting to: localhost:5900
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 599, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/local/lib/python2.7/dist-packages/websocket_server/websocket_server.py", line 136, in __init__
    StreamRequestHandler.__init__(self, socket, addr, server)
  File "/usr/lib/python2.7/SocketServer.py", line 655, in __init__
    self.handle()
  File "/usr/local/lib/python2.7/dist-packages/websocket_server/websocket_server.py", line 149, in handle
    self.read_next_message()
  File "/usr/local/lib/python2.7/dist-packages/websocket_server/websocket_server.py", line 161, in read_next_message
    b1, b2 = self.read_bytes(2)
  File "/usr/local/lib/python2.7/dist-packages/websocket_server/websocket_server.py", line 153, in read_bytes
    bytes = self.rfile.read(num)
  File "/usr/lib/python2.7/socket.py", line 384, in read
    data = self._sock.recv(left)
timeout: timed out
Segmentation fault (core dumped)
```

trunc8, 2021-04-08:
Hi @Jovian-Dsouza, can you make a new thread on the [forum](https://forum.jderobot.org/c/english/roboticsacademy/12)? I have faced multiple issues similar to yours. Let us try to jointly resolve it there.

Jovian-Dsouza, 2021-04-09:
Okay @trunc8  will do that

ReyDoran, 2021-06-15:
This problem is solved with new RADI 2.4.2

closed 2021-06-15 by ReyDoran