JdeRobot/RoboticsAcademy #836: [RADI] Update RADI to 0.3.14

pariaspe, 2021-04-08:
To include follow_turtlebot exercise

trunc8, 2021-04-08:
Hi @pariaspe, this is something I am keen to learn! Would I have the authorization to push the updated tag to RobotAcademy's DockerHub image?

jmplaza, 2021-04-08:
Hi @trunc8 , I'm afraid not. That is only for committers, long term RoboticsAcademy developers

trunc8, 2021-04-08:
Got it! I shall wait for my turn :)
I also suspected the RELEASE file at the root directory may be tied to an automation script that pushes the Docker image. Is that the case?

SakshayMahna, 2021-04-16:
Latest Tag: https://hub.docker.com/layers/jderobot/robotics-academy/latest/images/sha256-eb2cf69b0448f965607efdb193ea89a1d35deb677e985303f7a44690d491adee?context=repo

0.3.14 Tag: https://hub.docker.com/layers/jderobot/robotics-academy/0.3.14/images/sha256-eb2cf69b0448f965607efdb193ea89a1d35deb677e985303f7a44690d491adee?context=repo

closed 2021-04-16 by pariaspe