JdeRobot/RoboticsAcademy #843: Monitoring Gazebo Real-Time-Factor in real time and show it on the exercise webpage

jmplaza, 2021-04-09:
Right now we already monitor Brain frequency and GUI frequency (they both also allow setting their desired frequencies through a slider). Monitoring the "simulator frequency" too may provide information about the computation status of the running exercise. For instance, is the student's computer fast enough to run the exercise? is Gazebo's performance degrading and not reaching the Rea-Time-Factor of 1?

One possible implementation for this is in the exercise.py, just running the "gz" tool ("gz stats" command), processing the results and sending them to the browser through the websocket "ws_code" (the same used for sending the measured brain frequency and the measured GUI frequency).

trunc8, 2021-04-10:
Hi, please can one of the maintainers look at #844, in which I have tried to solve this issue?

@jmplaza and @SakshayMahna some of the threads in the exercise.py like measuring frequency and tracking real-time-factor run perenially. It might be more efficient to run them at intervals. What is your opinion on this? Would this problem get solved with Python multiprocessing?

Also, Sakshay is already working on the shift to multiprocessing architecture. Can I take more involvement in that task?

SakshayMahna, 2021-04-12:
Hello!
The measuring threads take up very small part of the computation. Maybe, running it in intervals won't make much difference. 

I already have completed my side of the work on multiprocessing architecture, you can have a look at #727, and suggest any changes or improvements.

jmplaza, 2021-04-25:
Hi @trunc8 , 

tracking real-time-factor inside exercise.py seems better for me than using Intervals at the web browser. As @SakshayMahna pointed out the take up very little computation. 

Even further, inside the browser the Intervals are dangerous on heavy load situations. In case of measuring frequencies in the browser I would choose Timeouts (creating the next Timeout at the very end of the current Timeout), so they would not accumulate in the case of heavy load situations.

ReyDoran, 2021-06-18:
This issue was solved on pull request [#1035 ](https://github.com/JdeRobot/RoboticsAcademy/pull/1035)

closed 2021-06-18 by ReyDoran