JdeRobot/RoboticsAcademy #845: Connection established popup needs to be clicked twice

trunc8, 2021-04-10:
The browser alert for "Connection established" or "Connection lost" requires the "OK" button to be clicked twice every time.
[image]

shreyas2000, 2021-04-10:
Could it be more of a feature than an issue , as it's conforming that connection to both websockets  :1905 and :2303 are established sucessfully . Although we can change it give a single alert after both connections are made.

trunc8, 2021-04-10:
Ahh, it's due to the two VNC connections. It's unlikely to be a feature, else the alert messages would be more verbose. It's also a little annoying for a user to click twice every time.

dgvsnl, 2021-04-11:
So what now, should we reduce it to one or make it verbose

closed 2021-05-10 by ReyDoran