JdeRobot/RoboticsAcademy #847: [Drone] Runtime Error when running drone Exercises

Jovian-Dsouza, 2021-04-11:
Exercise : Drone Gymkhana, Drone cat mouse (without web templates)

Currently there are some drone exercises which are not yet ported for web Templates . Running this exercises results in error as one module `qfi` needed for rqt_gui is not avialable.

I was able to solve this error by copying the `qfi` module from `JdeRobot/ThirdParty`
```bash
cd ~
git clone https://github.com/JdeRobot/ThirdParty
cd ~/ThirdParty/qflightinstruments/python
cp -r qfi /usr/lib/python2.7/dist-packages/
```

Should I do a PR resolving this issue in the RADI Dockerfile  or should I make a note on the respective exercise page. This might be helpful when porting these exercise to RADI

[image]

```bash
RosPluginProvider.load(rqt_drone_teleop/VelTeleop) exception raised in __builtin__.__import__(rqt_vel_teleop.vel_teleop, [VelTeleop]):
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 80, in load
    attributes['module_name'], fromlist=[attributes['class_from_class_type']], level=0)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_vel_teleop/vel_teleop.py", line 19, in <module>
    from sensorsWidget import SensorsWidget
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_vel_teleop/sensorsWidget.py", line 6, in <module>
    from qfi import qfi_ADI, qfi_ALT, qfi_SI, qfi_HSI
ImportError: No module named qfi

PluginManager._load_plugin() could not load plugin "rqt_drone_teleop/VelTeleop":
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/plugin_handler.py", line 102, in load
    self._load()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 55, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_gui_py/ros_py_plugin_provider.py", line 61, in load
    return super(RosPyPluginProvider, self).load(plugin_id, plugin_context)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 90, in load
    raise e
ImportError: No module named qfi
```

jmplaza, 2021-04-11:
Hi @Jovian-Dsouza , the exercises on previous releases of RoboticsAcademy are not supported anymore. This exercise is already available in RoboticsAcademy 2.3, the web-based release. Please migrate to 2.3. Dividing the development effort on providing support to previous releases takes energy away from improving the current one...

jmplaza, 2021-04-11:
AFAIK qflightinstruments is not required for exercises in 2.3 as the Python GUI has been replaced by a web GUI.

Jovian-Dsouza, 2021-04-11:
@jmplaza I thought this might helpful for people who would be migrating the drone exercises. 

plus this might be useful to those people who are not able to perform the exercise in RADI due t performance and bug issues.

closed 2021-04-11 by Jovian-Dsouza

iamarkaj, 2021-04-11:
Hi @Jovian-Dsouza 
What exact performance issues are you facing? Did you try this:
For Linux machines, GPU acceleration can be enabled by downloading Nvidia Container Runtime, as given [here](https://github.com/NVIDIA/nvidia-container-runtime)

Jovian-Dsouza, 2021-04-11:
Yes @iamarkaj  I have installed nvidia container runtime and all the nvidia drivers. 

I have share two screenshots , the first one is when running web template , note the FPS and the second one is also when running on the same RADI container. The only difference here is that I am manually starting the simulation using `roslaunch`.

[image]

[image]

I have already mentioned the performance issue on #833 
and on the forum : [issue-833-performance-issue-in-radi](https://forum.jderobot.org/t/issue-833-performance-issue-in-radi/147)

Jovian-Dsouza, 2021-04-11:
Also @jmplaza  @iamarkaj  as you can see form the screen shots most of the GUI components are missing in the web Template. For example the control sticks , take off and land button and sensor information. Implementing them as web gui is a better option but until then we can expose the Python GUI using noVNC

pariaspe, 2021-04-11:
Hi @Jovian-Dsouza, control sticks, takeoff and land buttons are not necessary to solve de exercise. Web Templates only contains the essential components to solve the exercise.

Jovian-Dsouza, 2021-04-11:
> Hi @Jovian-Dsouza, control sticks, takeoff and land buttons are not necessary to solve de exercise. Web Templates only contains the essential components to solve the exercise. 

True @pariaspe , we don't need these to solve the exercise but it does making debugging a bit easier.

pariaspe, 2021-04-11:
By now, we are not interested in adding them to the web template. Anyway, thanks for the suggestion

Jovian-Dsouza, 2021-04-11:
> By now, we are not interested in adding them to the web template. Anyway, thanks for the suggestion

Yes, for now lets work on adding more exercises and solving the performance issues. Currently I am experimenting on making the simulation utilise the Nvidia GPU