JdeRobot/RoboticsAcademy #859: [Follow_road_v2.3] errors in the migration devel branch

RUFFY-369, 2021-04-15:
I have pushed the migrated version of 'follow_road' to recreate & check for the error occuring during launching & connecting with exercise.html. If the error is resolved the migration can be further improved. Currently, I am trying to resolve the error so, it would be better if someone can also help me regarding it for faster migration
[image]

iamarkaj, 2021-04-15:
Hi @RUFFY-369 It is unable to find polaris_ranger_ev model which is included inside world file

iamarkaj, 2021-04-15:
Hi @RUFFY-369 it is unable to find polaris_ranger_ev which is included inside world file

pariaspe, 2021-04-27:
There are still a couple of things to do:

- [ ] Update to last RADI version.
- [ ] Add new models to drones assets.

RUFFY-369, 2021-04-28:
On it

closed 2021-05-04 by pariaspe