JdeRobot/RoboticsAcademy #876: Loading and Saving code from the web template

shashwat623, 2021-04-19:
I was recently going through the code base of the exercises running on the web template(color_filter). I found something strange, while loading  and saving the code from the in browser editor we are unnecessarily using the _**Exercise.py**_ application running on the server/container as an intermediary and thus increasing the number of websocket calls. 

To save the code written on the in template editor, we are sending the code to _**Exercise.py**_  **from**  _**ws_code.js**_  and then saving it on the host machine. 
 
// saveCode() function inside _ws_code.js_
`function saveCode(){
	// Get the code from editor and add header

	var python_code = editor.getValue();
	python_code = "#save" + python_code;
	console.log("Code Sent! Check terminal for more information!");
	websocket_code.send(python_code)   // websocket send call to exercise.py
}`

Similarly to load the code from host machine we are sending a load signal to _**Exercise.py**_  **from**  _**ws_code.js**_. Exercise.py is then uploading the code from host machine and **sending it back to**  _**ws_code.js**_

// loadCode function inside _ws_code.js_
`function loadCode(){
	// Send message to initiate load message
	var message = "#load";
	websocket_code.send(message);  // sending load code signal to exercise.py
}`

 _**ws_code.js**_  is running on the client side itself and doesn't need _**Exercise.py**_ running on server/container to handle load/save code requests for it. Aren't we just unnecessarily using extra websocket calls because of this? 
Should I go ahead and place a pull request with the required changes in place? Please have a look at this and let me know @dvalladaresv @jmplaza @SakshayMahna @pmvera 

Thank you!!

SakshayMahna, 2021-04-19:
Hello!

Nice observation. But, I think we are not using these currently to save and load the code. Since, we are saving and loading from the Docker Image, these files would eventually get deleted if the a running container is removed. In place of that, we are using Javascript functions to save/load them on/from our local computer. This way the files would remain persistent.

Probably, these functions were not deleted when we shifted to the other mechanism.

shashwat623, 2021-04-19:
Thanks for your reply Sakshay.
Yes, I believe you are correct as the `saveCode()` and `loadCode()` functions inside `ws_code.js` file **are no longer needed**, in fact after analyzing the code base of the application, I have come to a conclusion that these functions are never even even being called. They must have been used somewhere in the previous architectures and are of no use now. To load/save code to/from the web template the `local_functions.js` file is instead used as of now, which implements event listeners for any click on the load or save button on the web template. 
Also, like you said, saving and loading from the Docker Image wouldn't make sense as those files would be deleted when a running container is removed.
I was trying to understand the pre existing code base for the application as my proposed GSoC project idea for this summer is based on something similar i.e loading a pre trained Deep Learning model to the container and inferring the exercise outputs based on that. For this, understanding the pre existing pipeline would be essential. Such unnecessary functions which are no longer being used would just cause confusion in the future for people trying to understand the code base.

Should I go ahead and remove the above unused functions from `ws_code.js` and create a pull request for the same?

shashwat623, 2021-04-19:
Thanks for your reply @SakshayMahna 

Yes, I believe you are correct as the saveCode() and loadCode() functions inside ws_code.js file are no longer needed, in fact after analyzing the code base of the application, I have come to a conclusion that these functions are never even even being called. They must have been used somewhere in the previous architectures and are of no use now. To load/save code to/from the web template the local_functions.js file is instead used as of now, which implements event listeners for any click on the load or save button on the web template.
Also, like you said, saving and loading from the Docker Image wouldn't make sense as those files would be deleted when a running container is removed.

I was trying to understand the pre existing code base for the application as my proposed GSoC project idea for this summer is based on something similar i.e loading a pre trained Deep Learning model to the container and inferring the exercise outputs based on that. For this, understanding the pre existing pipeline would be essential. Such unnecessary functions which are no longer being used would just cause confusion in the future for people trying to understand the code base.

Should I go ahead and remove the above unused functions from ws_code.js and create a pull request for the same?

shashwat623, 2021-04-19:
Thanks for your reply @SakshayMahna 

Yes, I believe you are correct as the `saveCode()` and `loadCode()` functions inside `ws_code.js` file **are no longer needed**, in fact after analyzing the code base of the application, I have come to a conclusion that these functions are never even even being called. They must have been used somewhere in the previous architectures and are of no use now. To load/save code to/from the web template the `local_functions.js` file is instead used as of now, which implements event listeners for any click on the load or save button on the web template.
Also, like you said, saving and loading from the Docker Image wouldn't make sense as those files would be deleted when a running container is removed.

I was trying to understand the pre existing code base for the application as my proposed GSoC project idea for this summer is based on something similar i.e loading a pre trained Deep Learning model to the container and inferring the exercise outputs based on that. For this, understanding the pre existing pipeline would be essential. Such unnecessary functions which are no longer being used would just cause confusion in the future for people trying to understand the code base.

Should I go ahead and remove the above unused functions from `ws_code.js` and create a pull request for the same?

jmplaza, 2021-04-21:
Fixed in #879

closed 2021-04-21 by jmplaza