JdeRobot/RoboticsAcademy #885: [DroneCatMouse] Few issues

iamarkaj, 2021-04-21:
[x] The command console.print() doesn’t work.
[x] The ‘Stop’ and ‘Reset’ buttons don’t work properly either.

Original post can be found [here](https://forum.jderobot.org/t/feedback-about-drone-cat-and-mouse-exercise/158)

pariaspe, 2021-05-09:
Real Time Factor (RTF) should be added too. 

Updated exercises should look like:
[image]

pariaspe, 2021-05-09:
## Drone Cat Mouse
- [ ] The command console.print() doesn’t work.
- [ ] The ‘Stop’ and ‘Reset’ buttons don’t work properly either.
- [ ] Delete console messages after every re-submission of student's code
- [ ] Add RTF

[image]

## Follow Turtlebot
- [ ] The command console.print() doesn’t work.
- [ ] Delete console messages after every re-submission of student's code
- [ ] Add RTF

[image]

## Follow Road
- [ ] Code editor seems broken.
- [ ] Connection isn't established.
- [ ] The command console.print() doesn’t work.
- [ ] Delete console messages after every re-submission of student's code
- [ ] Add RTF

[image]

pariaspe, 2021-05-12:
Fixed in #944. Follow_road world might be revisited since gas_station model has been temporally disable cause an error while gazebo loading

closed 2021-05-12 by pariaspe