JdeRobot/RoboticsAcademy #902: Optimization of Gazebo execution inside the RoboticsAcademy Docker Image (RADI) for Windows (+Nvidia graphics card)

jmplaza, 2021-04-29:
It is now feasible: [https://devblogs.microsoft.com/commandline/the-initial-preview-of-gui-app-support-is-now-available-for-the-windows-subsystem-for-linux-2/#bonus-leverage-wsls-gpu-access-to-run-linux-applications-with-3d-acceleration](https://devblogs.microsoft.com/commandline/the-initial-preview-of-gui-app-support-is-now-available-for-the-windows-subsystem-for-linux-2/#bonus-leverage-wsls-gpu-access-to-run-linux-applications-with-3d-acceleration)

jmplaza, 2024-07-03:
Can we safely close this issue @dpascualhe , updating the instructions inside the [Robotics Academy UserGuide ](https://jderobot.github.io/RoboticsAcademy/user_guide/#windows) for Windows users to "Enable GPU Acceleration" ?

dpascualhe, 2024-11-05:
Bringing this issue back to life now that GPU acceleration for Linux is in a much more stable position (https://github.com/JdeRobot/RoboticsInfrastructure/issues/441)