JdeRobot/RoboticsAcademy #946: GUI.showImage() can not display one channel images

ReyDoran, 2021-05-13:
Running the following code on follow_line:

```
from GUI import GUI
from HAL import HAL
import cv2 as cv

HAL.motors.sendW(0.5)

while True:
    img = HAl.getImage()
    (b, g, r) = cv.split(img)
    ret, thresh = cv.threshold(r, 200, 255, cv.THRESH_BINARY)
    GUI.showImage(r)
```

Generates this error:
```
file “/RoboticsAcademy/exercises/follow_line/web-template/shared/image.py”, line 85, in add
metadata = MD(image.shape[0], image.shape[1], image.shape[2], byte_size)
IndexError: tuple index out of range 
```

Because grayscale images do not contain the third image.shape element that is trying to access ([source](https://docs.opencv.org/master/d3/df2/tutorial_py_basic_ops.html))

SakshayMahna, 2021-05-13:
Interesting error.

I can think of a good fix for it, but have to try it out.

In the `user_functions.py` file, we have the implementation for the `showImage` in `GUIFunctions`. Just before submitting the image to `shared_image.add(image)` function, we can apply something on the lines of `np.reshape(image, (image.shape[0],image.shape[1], -1) )`. This will reshape any form of image to 3 dimensions and the shared image would not crash the way it does.

closed 2021-05-20 by ReyDoran

ReyDoran, 2021-05-21:
In the end I opted for using `image = cv2.cvtColor(image, cv2.COLOR_GRAY2RGB)` as np.reshape wasn't reshaping correctly, but thanks for pointing me to the right direction!

SakshayMahna, 2021-05-21:
Okay! Great!