JdeRobot/RoboticsAcademy #958: RADI launch with ./start.sh not working with RADI-2.4.0-beta

ReyDoran, 2021-05-18:
The instruction 
`docker run -it -p 8000:8000 -p 2303:2303 -p 1905:1905 -p 8765:8765 -p 6080:6080 -p 1108:1108 jderobot/robotics-academy:2.4.0-beta ./start.sh`
generates an error while
`docker run -it -p 8000:8000 -p 2303:2303 -p 1905:1905 -p 8765:8765 -p 6080:6080 -p 1108:1108 jderobot/robotics-academy:2.4.0-beta bash`
and then executing `./start.sh` inside the container works.

trunc8, 2021-05-20:
TL;DR: ROS environment not getting sourced.

`./start.sh` with the shebang `#!/bin/bash` translates to `bash /start.sh`  
In the above command, `~/.bashrc` file does not automatically get sourced before executing the script. It is sourced at the time of opening a bash shell. This is crucial as the last line of the docker image's `~/.bashrc` file is `source /opt/ros/melodic/setup.bash`

ReyDoran, 2021-06-01:
Hi, thanks for the info. Do you know how can I source it properly? I'm not sure how to approach this.

trunc8, 2021-06-01:
A possible way to get around this problem would be including `source /opt/ros/melodic/setup.bash` line in the `start.sh` script.  
I'm unsure what would be considered as best practice in this case though.

ReyDoran, 2021-06-02:
Thanks it worked. At least temporarily while we are testing the accelerated RADI we can go with this fix.

ReyDoran, 2021-06-02:
Issue solved in commit [98678c511ec262ff87f91edd4df227df8289ad5e](https://github.com/JdeRobot/RoboticsAcademy/pull/985/commits/98678c511ec262ff87f91edd4df227df8289ad5e)

closed 2021-06-02 by ReyDoran