JdeRobot/RoboticsAcademy #1001: Solve connectivity issues with RADI 2.4 versions

davidrol6, 2021-06-07:


ReyDoran, 2021-06-08:
I was not able to connect on exercises Follow Turtlebot and Drone Cat Mouse when using Windows. Are there any other issues?

iamarkaj, 2021-06-08:
> I was not able to connect on exercises Follow Turtlebot and Drone Cat Mouse when using Windows. Are there any other issues?

I suppose the problem lies because of the changes([1](https://github.com/JdeRobot/RoboticsAcademy/blob/38c5efc31d58d9582c2a26df3228f6e08878405b/exercises/static/exercises/drone_cat_mouse/web-template/assets/ws_code.js#L91),[2](https://github.com/JdeRobot/RoboticsAcademy/blob/38c5efc31d58d9582c2a26df3228f6e08878405b/exercises/static/exercises/drone_cat_mouse/web-template/assets/ws_code.js#L93)) made in ws_code.js file of the drone exercises in [this](https://github.com/JdeRobot/RoboticsAcademy/commit/7e567c6bebe831b2d67527ca1fbb7ded794db805) commit.

pariaspe, 2021-06-12:
Hi @ReyDoran, 
I've tested drone exercises on Linux with RADI-2.4.2 and it went ok. That's the error persists in the lastest 2.4 RADI version on Windows?

ReyDoran, 2021-06-15:
It does not happen anymore with RADI 2.4.2

closed 2021-06-15 by ReyDoran