JdeRobot/RoboticsAcademy #1028: [noetic] Migrate remaining exercises to RADI 3.1

pariaspe, 2021-06-15:
- [ ] Color Filter
- [ ] Reconstruction 3d
- [ ] Car Junction
- [ ] DL-Digit Classifier

ReyDoran, 2021-06-22:
On 3D Reconstruction, I am getting this error 
`Exception in thread Thread-15:
Traceback (most recent call last):
 File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner self.run()
 File "/usr/lib/python3.8/threading.py", line 870, in run self._target(*self._args, **self._kwargs)
 File "/RoboticsAcademy/exercises/3d_reconstruction/web-template/exercise.py", line 154 in process_code
    exec(iterative_code, reference_environment)
 File "<string>", line 25, in <module>
TypeError: slice indices must be integers or None or have an _index_ method`
when running the following sample code (works on 2.4.2):
```
from GUI import GUI
from HAL import HAL
import cv2
import numpy as np
import math
# Enter sequential code!
block = 23
tam = (block-1)/2
height = 0
width = 0
channel = 0
position_camleft = []
position_camright = []
vez = 0
while True:
    if(vez == 0):
        position_camleft = HAL.getCameraPosition('left')
        position_camright = HAL.getCameraPosition('right')
        p_total = []
        # Enter iterative code!
        imageLeft = HAL.getImage('left')
        imageRight = HAL.getImage('right')
        
        [height,width,channel] = imageLeft.shape
        kernel = np.ones((1,1),np.uint8)
        imgleft = np.copy(imageLeft)
        imgright = np.copy(imageRight)
        hsv_left = cv2.cvtColor(imgleft, cv2.COLOR_RGB2HSV)
        hsv_right = cv2.cvtColor(imgright, cv2.COLOR_RGB2HSV)
        gray_left = cv2.cvtColor(imgleft, cv2.COLOR_RGB2GRAY)
        gray_right = cv2.cvtColor(imgright, cv2.COLOR_RGB2GRAY)
        edgesleft = cv2.Canny(gray_left,50,120,5)
        edgesright = cv2.Canny(gray_right,50,120,5)
        edgesright = cv2.morphologyEx(edgesright, cv2.MORPH_CLOSE, kernel)
        edgesleft = cv2.morphologyEx(edgesleft, cv2.MORPH_CLOSE, kernel)
        edgesleft_img=cv2.cvtColor(edgesleft,cv2.COLOR_GRAY2BGR)
        edgesright_img=cv2.cvtColor(edgesright,cv2.COLOR_GRAY2BGR)
        [alto,ancho] = edgesleft.shape
        [y_left,x_left] = np.where(edgesleft[tam:height-tam,tam:width-tam]);
        
        y_left = y_left[0:len(y_left):3] + tam
        x_left = x_left[0:len(x_left):3] + tam
        for aux in xrange(len(x_left)):
            pointIn_ini = np.array([x_left[aux],y_left[aux],1])
    
            pointInOpt_ini = HAL.graficToOptical('left',pointIn_ini)
            point3d_ini = HAL.backproject('left',pointInOpt_ini)
    
            punto3d = [point3d_ini[0],point3d_ini[1],point3d_ini[2]]
            centro = [position_camleft[0], position_camleft[1], position_camleft[2]]
    
            rayo_project = np.cross(punto3d,centro)
            p_aux = [(punto3d[0]-centro[0])/2+centro[0],(punto3d[1]-centro[1])/2+centro[1],0]
            p_aux[2] = (-rayo_project[0]*p_aux[0]-rayo_project[1]*p_aux[1])/rayo_project[2]
            p_aux3d = np.array([p_aux[0],p_aux[1],p_aux[2],1])
      
  
            projected1 = HAL.project('right', point3d_ini)
            projected2 = HAL.project('right', p_aux3d)
[... 78 lines cut ...]
                    mua = float(numer)/float(denom);
                if(d4343 == 0):
                    mub = 0
                else:
                    mub = float((d1343 + d4321 * (mua)))/float(d4343);
                pa = [0,0,0]
                pb = [0,0,0]
                pa[0] = position_camleft[0] + mua*p21[0];
                pa[1] = position_camleft[1] + mua*p21[1];
                pa[2] = position_camleft[2] + mua*p21[2];
                pb[0] = position_camright[0] + mub*p43[0];
                pb[1] = position_camright[1] + mub*p43[1];
                pb[2] = position_camright[2] + mub*p43[2];
                if (math.sqrt((pb[0]-pa[0])**2+(pb[1]-pa[1])**2+(pb[2]-pa[2])**2)>0.4):
                    error=1
                p_medio = [0,0,0]
                p_medio[0] = (pa[0] + pb[0])/2
                p_medio[1] = (pa[1] + pb[1])/2
                p_medio[2] = (pa[2] + pb[2])/2
                punto3D_final = HAL.project3DScene(p_medio)
                punto_pintado = [[(punto3D_final[0]), (punto3D_final[1]), (punto3D_final[2]), int(color[0]),int(color[1]),int(color[2])]]
                if (error == 0 and p_medio[2]<4700 ):
                    GUI.ShowNewPoints(punto_pintado)
        vez = 1
```

pariaspe, 2021-06-24:
Car Junction exercise is working but it has some warnings trying to load all models (see #1051).

ReyDoran, 2021-07-02:
3D Reconstruction is working. The problem I posted below was due to the change from Python 2 to Python 3. The problems were:
- xrange() function was removed in Python3 (we can use range() as an alternative)
- some variable type inference changes
I'm removing sample code as it is close to be the exercise solution.
I think this issue can be closed now.

closed 2021-07-02 by pariaspe