JdeRobot/RoboticsAcademy #1073: [RADI] update to tag 3.1.2

ReyDoran, 2021-06-30:
This release includes:
- Tested support for exercises: follow line, obstacle avoidance, vacuum cleaner, vacuum cleaner loc, color filter, car junction

pariaspe, 2021-06-30:
I would also add:
- Tested support for exercises: position_control, rescue_people, labyrinth_escape, follow_road, follow_turtlebot

jmplaza, 2021-06-30:
11 tested exercises! Awesome!

pariaspe, 2021-07-01:
Tag [3.1.2](https://hub.docker.com/layers/jderobot/robotics-academy/3.1.2/images/sha256-e9ef7359739b11a323a2059b029d1fcef4f4ddfb1bd39a326c8c2d589ff2e4c4?context=explore) released

closed 2021-07-01 by pariaspe

ReyDoran, 2021-07-01:
I had to reupload it, new link: [tag 3.1.2](https://hub.docker.com/layers/jderobot/robotics-academy/3.1.2/images/sha256-59b11e6d9b761d6e9d995c129acf5732a28e9c0a54304d615e5e895adc31468e?context=explore)