JdeRobot/RoboticsAcademy #1079: Error while running the noetic image container

shashwat623, 2021-07-01:
I get an error while starting a container for the latest noetic docker image. It seems it cannot locate the `start.sh` file from the mount point...

[image]

pariaspe, 2021-07-01:
You can run it with:
```
docker run -it --device /dev/dri -p 8000:8000 -p 2303:2303 -p 1905:1905 -p 8765:8765 -p 6080:6080 -p 1108:1108 jderobot/robotics-academy:3.1.1 ./start-3.1.sh
```

shashwat623, 2021-07-01:
Thanks @pariaspe . Should we change the noetic readme too then? It mentions `start.sh`

pariaspe, 2021-07-01:
We'll better change script name

shashwat623, 2021-07-01:
Alright @pariaspe

closed 2021-07-01 by shashwat623