JdeRobot/RoboticsAcademy #1089: Simulation control to manager

SakshayMahna, 2021-07-02:
Currently, the `exercise.py` file handles both the simulation and the code websocket. 

To restructure the code in such a way:

- `manager.py` handles the simulation control
- `exercise.py` handles the code websocket

SakshayMahna, 2021-07-12:
A proper transfer to `manager.py` would require certain changes to the main template `base.html`. Therefore, affecting all the exercises. Hence, the steps to be followed are:

- Make bare changes to Follow Line exercise
- Extend to other exercises
- Apply proper refactoring and the changes to `base.html`

ReyDoran, 2021-07-13:
Hi, may I help with this issue? Migrating this changes to noetic branch, or applying refactoring to base.html for example?

SakshayMahna, 2021-07-13:
Hello, yes that would be great!

For the simulation control, we are going to have 3 commands (resume, pause and reset). Currently only the reset is being used, and we may later extend the use for resume and pause as well.

Currently, the code, gui and manager websockets are mainly controlled by functions in `ws_code.js`. For instance, the `resetSim()` function, calls functions present in `ws_gui.js` and `launcher.js`. I was thinking that we should abstract this functionality into a different file like `controller.js` (not sure of the name). Then, all the different functions in `ws_code`, `ws_gui` and `launcher` could be called from `controller`.

In addition, this would keep the code cleaner when we introduce the pause and stop functionality later on.

It would be great to get some comments or suggestions, this looks like a good way to go, right? I'll add this functionality in the `main` branch, then you may extend it to the `noetic` branch.

ReyDoran, 2021-07-13:
Yes, the `controller.js` file sounds good to me!
I will start migrating the current improvements to noetic, and will continue when the rest is ready.
When I finish I can also take a look at the resume and pause commands, if you are now working now on the `controller.js` file.

SakshayMahna, 2021-07-13:
Great! I'll add a PR for the controller.

SakshayMahna, 2021-07-14:
Hello @ReyDoran 

I have added the `controller.js` file. The main changes are:

- Create a new file `controller.js` . Remove the unnecessary components from `ws_code.js`, mainly in the functions `submitCode()`, `stopCode()`. Remove the function `resetSim()` and the variable `running`.
- Add `controller.js` script to `exercise.html` and rename the functions that are called in `base.html`.

I have added the functions for start and stop simulation. But, they don't yet call the manager for the simulation controls.

ReyDoran, 2021-07-14:
I have a [pr-1137](https://github.com/JdeRobot/RoboticsAcademy/pull/1137) which adds stop and resume functions to manager.py on follow line. I'm going to migrate it to all the remaining exercises, but I'd like your approval before merging it to this branch and to master.

ReyDoran, 2021-07-18:
Closed by #1147 , pushed to noetic branch

closed 2021-07-18 by ReyDoran