JdeRobot/RoboticsAcademy #1096: [RADI] Connection closed automatically after a while of the stablishment

pariaspe, 2021-07-03:
Connection get closed due to a Timeout Error. This screenshot was taken using RADI v3.1.2

[image]

jmplaza, 2021-07-03:
It is weird to see Python-2.7 in SocketServer, maybe we should upgrade to its Python-3.X release. 

RADI  3.1 works with Ubuntu 20.04 as underlying linux distribution (FROM nvidia/opengl:1.2-glvnd-runtime-ubuntu20.04) and ROS Noetic, which does work with Python 3.X.  There is no need to use Python-2.7 for anything.

In addtion, in the RADI-3.1.2 docker file I see
# websocket server dependency
RUN pip3 install websocket_server posix-ipc django
RUN python3.8 -m pip install websockets asyncio

but from the snapshot you attached it seems it is using websocket server with Python-2.7...

iamarkaj, 2021-07-04:
[image]

I get the same timeout error after 30 seconds but the python version is different for me in RADI-3.1.2

closed 2021-07-14 by pariaspe