JdeRobot/RoboticsAcademy #1127: [gh-pages] Creation of exercise page "Amigo Bot"

JuanManuelCarretero, 2021-07-11:
As development on the exercise Amigo bot is getting nearly finished, the page and resources for the exercise are being made and updated to make sure that the exercise can be launched as soon as it is ready for testing

jmplaza, 2021-07-11:
There will be more exercises with AmigoBot robot, let's call this exercise "Laser mapping exercise"

ReyDoran, 2021-07-26:
Closed by #1181

vladkrav, 2021-07-28:
Information to upload to the Laser mapping exercise website:

**Application Programming Interface**
from HAL import HAL - to import the HAL(Hardware Abstraction Layer) library class. This class contains the functions that sends and receives information to and from the Hardware.
from GUI import GUI - to import the GUI(Graphical User Interface) library class. This class contains the functions used to view the debugging information, like image widgets.

**Sensors and robot state**
HAL.getPose3d().x to get position x of the robot.
HAL.getPose3d().y to get position y of the robot.
HAL.getPose3d().yaw to get the orientation of the robot.
HAL.motors.sendW() to set the angular velocity.
HAL.motors.sendV() to set the linear velocity.
 HAL.getLaserData() to get the data of the LIDAR.
HAL.getSonarData_0() to get the sonar data of the sonar 1. 
HAL.getSonarData_1() to get the sonar data of the sonar 2. 
HAL.getSonarData_2() to get the sonar data of the sonar 3. 
HAL.getSonarData_3() to get the sonar data of the sonar 4. 
HAL.getSonarData_4() to get the sonar data of the sonar 5. 
HAL.getSonarData_5() to get the sonar data of the sonar 6. 
HAL.getSonarData_6() to get the sonar data of the sonar 7.
HAL.getSonarData_7() to get the sonar data of the sonar 8.

**Theory**
_Mapping with known positions_
Mapping with known positions assumes that the current position of the robot is known. This technique consists of converting the distance measurements of the different laser beams into Cartesian coordinates relative to the robot. The distance of the beams reflects the existence of an obstacle; therefore, these Cartesian coordinates are inserted reflecting obstacles in an occupation grid with respect to the current position of the robot.
This technique is not entirely real because in most cases, the position of the robot is unknown. Therefore, other techniques such as SLAM are used.

_Occupancy grid_
An occupation grid is a discretization of the robot's environment in cells. This discretization will be given by the size of the world in which the robot is located. With an occupation grid, a matrix is handled whose cells will contain a probability value, which indicates the certainty that in that position there is an obstacle (1), there is free space (0), or it has not been explored for the moment (gray space).
The occupation grids were initially proposed in 1985 by Moravec and Elfes. The biggest advantage of these types of maps is that they are easy to build and maintain, even in large environments. Also, it is easy for a robot to determine its position within the map just by knowing its position and orientation, since the geometry of the cells corresponds to the geometry of the terrain.
On the other hand, the basic problem with this type of map is the large amount of memory required for storing the information.

An example of a map obtained with the Mapping technique with known positions. 
[image]

References:
[Alberto Elfes](https://www.cs.cmu.edu/~motionplanning/papers/sbp_papers/integrated4/elfes_occup_grids.pdf)

jmplaza, 2021-07-29:
The structure is fine but the contents are not related to laser mapping exercise yet

ReyDoran, 2021-07-30:
Closed by #1198

closed 2021-07-30 by ReyDoran