JdeRobot/RoboticsAcademy #1165: Improve the connection protocol for PX4-SITL over drone exercises for multi-drone exercises

iamarkaj, 2021-07-18:
Only for Drone Cat Mouse

RUFFY-369, 2021-08-25:
Hi @iamarkaj  , are you currently working on this task for the enhancement?

iamarkaj, 2021-08-25:
> Hi @iamarkaj , are you currently working on this task for the enhancement?

Hi @RUFFY-369 Not really but I am eager to collaborate if there is a way out

iamarkaj, 2021-08-30:
Hello @RUFFY-369 

For all single-drone exercises the launching order is `gazebo -> px4 -> mavros`
For drone-cat-mouse exercise it is `gazebo -> px4_cat -> mavros_cat -> px4_mouse -> mavros->mouse`

**PX4 Test** : `commander check` is used to run preflight checks. If `Prearm check: OK` it will continue to launch mavros else it will wait. 

`commander check` does not require vehicle-id or vehicle-name. For other single-drone exercises this works fine but for drone-cat-mouse exercise `commander check` cannot be used to test px4_mouse. Currently, it waits for 5 seconds ([here](https://github.com/JdeRobot/RoboticsAcademy/blob/e8aae0e640b9f4286cc1461ea1e38c8b97302ed2/exercises/static/exercises/drone_cat_mouse/web-template/launch/launch.py#L85)) assuming the px4_mouse will be launched. But this needs to be improved.

For info on how I think this be solved is mentioned [here](https://theroboticsclub.github.io/gsoc2021-Arkajyoti_Basak/blog/2021-07-02-Week-4/)

RUFFY-369, 2021-09-02:
Hi @iamarkaj , if I am getting it correctly then you are saying that the commander check happens just after 'px4_cat' in `gazebo -> px4_cat -> mavros_cat -> px4_mouse -> mavros_mouse` and doesn't happens in subsequent px4 launches of remaining single/multiple drones?
Also, is commander check able to utilise the remote port assigned to each vehicle?

pariaspe, 2021-09-03:
Hi @RUFFY-369 , Yes, `commander` checks if px4 is running, so it happens right after "px4_cat" as you pointed. The main problem is that we can't distinguish different px4 from different drones. We didn't find a way to assign ports to each px4 instance.

RUFFY-369, 2021-09-04:
Hi @pariaspe , isn't `SITL_UDP_PRT` parameter used to set the UDP port on the PX4 SITL side? Can't we utilise that?
And also does `commander check` only runs for single PX4 instance once? How does it performs the preflight checks, I mean how does it identifies the vehicle?

pariaspe, 2021-09-05:
Hi @RUFFY-369 , 
We are using PX4 [modules](https://dev.px4.io/master/en/middleware/modules_main.html) to check the readiness state. We decide to use `commander check` ([ref](https://dev.px4.io/master/en/middleware/modules_system.html#commander)) but any other may be used too. 
By now, we didn't find a way to link UDP port to each PX4 instance or vehicle.

RUFFY-369, 2021-09-07:
Hi @pariaspe , does this (micrortps_agent) can help for the linkage or communication? (below)
[image]
https://docs.px4.io/master/en/simulation/multi_vehicle_simulation_gazebo.html#build-and-test-rtps-dds

pariaspe, 2021-09-09:
Hi @RUFFY-369 , not sure if that is what we are looking for, but it might work. What you propose to do with that connection? 
Feel free to try your proposals and share your findings.

RUFFY-369, 2021-09-10:
Hi @pariaspe , I found it important to share 'cause as you said you guys didn't find a way to link UDP port to each PX4 instance or vehicle. And here as these guys have mentioned that through RTPS insterface each vehicle instance is allocated a unique MAVLINK sys id to communicate with unique UDP port. So, maybe if we can link some way to check the readiness of the vehicle state.
I haven't went into that depth to finalise a method so, I have to look into it more.

pariaspe, 2021-09-11:
Hi guys @iamarkaj @RUFFY-369 ,
I've been playing with `micrortps_agent` and I didn't find it really interesting for us. 

However, I find something interesting related with `commander` module. There is a test that checks the commander status, it output shows something like:
```
parias@parias-msi:~/repos/Firmware/build/px4_sitl_default/bin$ ./px4-commander_tests 
INFO  [commander_tests] RUNNING TEST: armingStateTransitionTest
WARN  [commander] Transition denied: ARMED to INIT
WARN  [commander] Transition denied: ARMED to SHUTDOWN
WARN  [commander] Transition denied: STANDBY_ERROR to STANDBY
WARN  [commander] Transition denied: IN_AIR_RESTORE to STANDBY
WARN  [PreFlightCheck] Arming denied! Press safety switch first
INFO  [commander_tests] TEST PASSED: armingStateTransitionTest
INFO  [commander_tests] RUNNING TEST: mainStateTransitionTest
INFO  [commander_tests] TEST PASSED: mainStateTransitionTest
INFO  [commander_tests] ALL TESTS PASSED
INFO  [commander_tests]   Tests passed :      2
INFO  [commander_tests]   Tests failed :      0
INFO  [commander_tests]   Tested assertions : 138
```
We might just consider to check if any test failed or not, what do you think?

RUFFY-369, 2021-09-24:
HI @pariaspe , apologies for the late response. Is there any way which was there for this test to differentiate between the different instances of vehicles?
Is this the script which  is being used for the tests: [script](https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/commander/commander_tests/state_machine_helper_test.cpp)

pariaspe, 2021-09-24:
Yes @RUFFY-369, those are the source files. I was referring to the already built ones.

RUFFY-369, 2021-09-24:
Yeah, @pariaspe , are these tests happening for all the vehicle instances?

pariaspe, 2021-09-24:
Yes, @RUFFY-369, they are

pariaspe, 2021-11-04:
Basic method implemented in #1335

jmplaza, 2021-11-21:
May we close this issue @pariaspe ?

pariaspe, 2021-11-21:
Connection protocol has been improved with a simple checker and a better method might be interesting. Anyway, we can close this and open a new issue on the future for this

closed 2021-11-29 by pariaspe