JdeRobot/RoboticsAcademy #1396: Regular pressing of  backspace

RUFFY-369, 2021-11-18:
This is regarding a small functionality in the web-template based exercises.
If someone can confirm me regarding this, i.e., while deleting in console one has to delete by pressing backspace multiple times instead of just keeping it pressed (like how it normally happens).
Is it something on my side or is it common & happens within the web server?

ShreeSinghi, 2023-01-16:
I am also facing the same issue on my machine. If no one else is working on this issue could I work on it? 

> This is regarding a small functionality in the web-template based exercises. If someone can confirm me regarding this, i.e., while deleting in console one has to delete by pressing backspace [...]

MashiatK, 2023-02-08:
I want to work on this. If the issue is open, please assign it to me.

jmplaza, 2023-02-08:
HI @MashiatK , there is no need to explicit assignment, just work on it and feel free to submit a Pull Request solving it :-)

Rachit-Gandhi, 2023-04-05:
I will be taking a look into the same, just putting here if someone wants to join in.

Surajpatra700, 2023-12-24:
I want like to work on this issue. Is the issue open ?

jmplaza, 2023-12-25:
@MashiatK  @Surajpatra700  @Rachit-Gandhi No permission is needed to work on this issue. 

Just use RoboticsAcademy following the documentation, get familiar with it.

After that, try to reproduce the problem reported in the issue description. Locate the console in the exercise frontend. I would suggest to start with the FollowLine exercise. And then, if the problem still is there, just work on the solution. Maybe it has been solved in the last releases of RoboticsAcademy as this issue in two years old...

closed 2024-07-29 by javizqh