JdeRobot/RoboticsAcademy #1425: Fix visualization error in Laser Mapping

ReyDoran, 2021-11-30:


ReyDoran, 2021-11-30:
After the first pr #1426 it does only happen on accelerated simulation

Naveen175py, 2023-02-24:
Hey I want to work on this issue please assign me this issue

Surajpatra700, 2024-02-21:
@ReyDoran Can you please tell me which specific file is showing this issue ?

Artinong, 2025-01-16:
hey @ReyDoran i want to solve this issue can you provide me the pull request

JuanFernandezEva, 2025-01-17:
Hi @Artinong, this is an issue for an old version of the laser mapping exercise, I'm currently working on this exercise and updating it to the current version of RoboticsAcademy, it shouldn't be a problem anymore.

dduro2020, 2025-03-31:
When changing the map the image on the right (generated map) should be reset. It updates correctly when requested, but does not reset at start
.

dduro2020, 2025-04-07:
There seems to be a problem with the universe changing when the exercise is running. I'm still not 100% sure if that's the reason, but it's quite likely. This error causes the docker to die without being able to run any other exercises

closed 2025-04-08 by javizqh