JdeRobot/RoboticsAcademy #1520: Unable to run RADI with GPU acceleration

pariaspe, 2022-01-23:
I'm getting this error with VGL when I'm trying to run the latest RADI with GPU acceleration:
```
[VGL] ERROR: Could not open display /dev/dri/card0.
[WARN] [1642851485.742479]: [GAZEBO] Check timeout exceeded
```

Since VGL fails, Gazebo launching also fails and I can not play with the exercise.
I was able to use GPU acceleration with RADI 3.1 versions.

soumilbaldota, 2022-03-14:
You could try to do this and see if it works.
```sudo nvidia-xconfig -a --allow-empty-initial-configuration --use-display-device=None \ --virtual=1920x1200 --busid {busid}```

soumilbaldota, 2022-03-14:
Similar to 
[link](https://github.com/VirtualGL/virtualgl/issues/108#issuecomment-531851472)

soumilbaldota, 2022-03-14:
Similar to (https://github.com/VirtualGL/virtualgl/issues/108#issuecomment-531851472)

pariaspe, 2022-03-18:
Hi @soumilbaldota, thanks for the suggestion but it didn't work in my system. Have you tried it?

pariaspe, 2023-01-09:
Solved in #1902. Thanks @iamarkaj !

closed 2023-01-09 by pariaspe