JdeRobot/RoboticsAcademy #1531: Improve documentation for developers

jmplaza, 2022-01-28:
explaining the main RoboticsAcademy components, how to deploy it locallly, etc...

pariaspe, 2022-01-31:
RA architecture section is great (the [image](https://github.com/JdeRobot/RoboticsAcademy/blob/fa163fc484a1266011e6d868a0e26bfbe5da98b1/README.md?plain=1#L88) is amazing), but I miss some explanation about it. 
- What is each websocket used for? Manager websocket? 
- How are they used? The current explanation is correct, but i suggest to tabulate explanation to better readability and understanding.
- A couple of websockets (code+GUI) is used for each robot not each exercise. Explain the used port choosing convention.
- As part of RAMP protocol, each msg send through websockets uses its five first characters as message definition. But then, why are messages like *#image* or like *#map*.
- I also miss some info about repo structure. It also requires some cleaning (issue #1530).

What about moving all this info to Github Wiki page? It can also be extended in the future with extra information (dependencies, etc.)

pariaspe, 2022-02-05:
Some more suggestions:
- I would add some kind of index to navigate into the README because it's getting bigger. Or even we can explore to split current README into several markdown files.
- I miss a list with the ports used.
- I would transform the RAMP protocol info into a table for better readability. Something like:

| Command | Field | Description | 
| --- | --- | --- |
| ... | ... | ... |

- Repository structure (instead of "architecture") is incomplete. We have two kind of branches, source code ones ([`melodic`](https://github.com/JdeRobot/RoboticsAcademy/tree/melodic), [`noetic`](https://github.com/JdeRobot/RoboticsAcademy), [`foxy`](https://github.com/JdeRobot/RoboticsAcademy/tree/foxy)) and documentation one ([`gh-pages`](https://github.com/JdeRobot/RoboticsAcademy/tree/gh-pages)). I would add the links to each branch as I've done.
- Noetic (master) branch structure is incomplete. I would also add some graphic representation.
```
RoboticsAcademy
├─ manage.py
├─ academy
├─ docs
├─ exercises
│        settings.py
│        urls.py
│        wsgi.py
│        __init__.py
└─requirements.txt
```
- I also miss some text formatting. Adding bold text for keywords, etc.

felicidadaqm, 2022-02-06:
We could add some brief explanation about the manager websocket as @pariaspe suggested, althougth most information about this websocket is spread in other parts of the documentation.

_`"The manager websocket is in charge of requesting the exercises and handling the control of the simulation. For example, it starts/stop/pause/resume/kill the Gazebo simulation. It also starts the VNC server. The code written by the user is sent first from the browser to the manager.py proccess through the manager websockets. Then, the manager.py checks the code with Pylint and returns the result to the browser"`_

Feel free to edit, add information about the ports used or anything else as I may lack some deeper understanding of the code yet.

Also, I noticed repetitions of the .html files for some exercises (color_filter or follow_line, for example). I'll leave the routes in which you can find these files:

```
RoboticsAcademy
├─ exercises/templates/exercises/{{exercise_name}}/exercise.html
├─ exercises/statis/exercises/{{exercise_name}}/web-template/exercise.html
```

They are not exactly the same code, so I'm not sure if both of them are used, or if one of them it's just an old version of the .html. Are both of them used? If so, maybe we should add a little explanation about why these files are duplicated.

Finally we could add a little information about the rest of the branches too, something similar to what we have in the unibotics-exercises documentation:

`"The rest of the branches will have the designation 'issue-xxxx', being xxxx the name correspondent to the incidence they are attached to. This number is given automatically by GitHub when an issue is open"`

felicidadaqm, 2022-02-13:
I made some small changes in the linked pull request, such as:

1. Links to the branches
2. Information about the manager websocket
3. Information about the issues branches

I'll ask about the things I have no context of and keep updating the documentation in the following weeks

closed 2022-03-28 by dvalladaresv