JdeRobot/RoboticsAcademy #1561: [Obstacle avoidance] Front-end refactorization

pariaspe, 2022-03-04:
Hey, we could start identifying components over the actual web and list them here

[image]

DanielinR, 2022-03-07:
Here is an excel document that identify some of the components i found.
[image]

DanielinR, 2022-03-08:
I've created a first version of the Obstacle avoidance exercise refactorized in the branch issue-1561. The most important changes are ubicated in the exercise.html and in the tamplates folder.
[image]

dvalladaresv, 2022-03-10:
In the branch issue-1561-review I have added an example of building the exercise template using the assets field of the exercise template, with the idea of not having n templates for the exercises.