JdeRobot/RoboticsAcademy #1570: [frontend] vertical sliding separator in exercise webpages

jmplaza, 2022-03-10:
Let's put it in the new base2.html, so the user may select more space to the web text editor or to the right side of the web page of the exercise... We want something similar to this example at overleaf.com:

https://user-images.githubusercontent.com/10961573/157725072-292046ec-b765-4c5f-8916-c109dd05e69b.mp4

DanielinR, 2022-03-11:
At this moment the robotics exercises work like this:

https://user-images.githubusercontent.com/60990208/157857853-ee70043e-5f38-4323-904d-9266c50cfe2e.mp4

Which improvement should we focus on?

jmplaza, 2022-03-11:
Maybe on doing the widgets more "responsive"...

aPR0T0, 2023-02-22:
I would like to work on this

jmplaza, 2023-03-05:
Hi @aPR0T0 , just install RoboticsAcademy, use it and submit your Pull-Request for solving this issue. Don't focus on the classic exercises, just on the brand new FollowLine flavour, which is based on REACT. We are migrating all the exercises to the new REACT-based design of the frontend.

closed 2023-06-06 by ango1994