JdeRobot/RoboticsAcademy #1605: [Autoparking] improve the car used in this exercise

jmplaza, 2022-03-25:
to follow Ackermann steering. Related to [https://github.com/JdeRobot/CustomRobots/issues/145](https://github.com/JdeRobot/CustomRobots/issues/145) and [https://github.com/JdeRobot/CustomRobots/issues/149](https://github.com/JdeRobot/CustomRobots/issues/149)

BhaveshAnandpara, 2022-07-27:
Hey ! @antoniojromero are you working on this Issue ? I would like to contribute also!

jmplaza, 2022-07-27:
Hi @BhaveshAnandpara , we are almost done with this issue, @akshaynarisetti has done a great job with it. Take a look at the latest video https://twitter.com/JdeRobot/status/1549634989255725057 from the last week. More info on this is available at [Akshay's GSoC page](https://theroboticsclub.github.io/gsoc2022-Akshay_Narisetti/) and [current week report](https://theroboticsclub.github.io/gsoc2022-Akshay_Narisetti/2022/07/24/Coding-Period-Week-8.html).

It will be soon available at [CustomRobots repository](https://github.com/JdeRobot/CustomRobots) and integrated in the autoparking exercise of RoboticsAcademy

jmplaza, 2022-07-27:
Help with be needed @BhaveshAnandpara for creating nice and light Gazebo scenarios for this AutoParking exercise and for betatesting the exercise with the new car model :-)

GauravKumar9920, 2023-01-07:
Hey, anyone working on this issue I would like to contribute on this issue.

jmplaza, 2023-01-08:
HI @GauravKumar9920 ,
yes, @Amglega almost has solved it, take a look at his [demo video](https://www.youtube.com/watch?v=CucqZrXDOto) at JdeRobot's YouTube channel regarding his solution.

PR #1908 (2023-02-05): added autopark_v2 exercise
Added new exercise based on the autopark one. The main improvement is the Ackerman geometry for the car. The new model has a LIDAR sensor also.

Resolves #1605
closed 2023-02-05 by dvalladaresv