JdeRobot/RoboticsAcademy #1669: Laser Mapping docker run command not working

Apoorvgarg-creator, 2022-04-13:
[Laser Mapping Exercise](http://jderobot.github.io/RoboticsAcademy/exercises/MobileRobots/laser_mapping) the command mentioned in the heading "How to perform the exercise ?"
```shell
docker run --rm -it -p 8000:8000 -p 2303:2303 -p 1905:1905 -p 8765:8765 -p 6080:6080 -p 1108:1108 jderobot/robotics-academy:3.1.4 ./start-3.1.sh
``` 
 Error -
[image]

Solution -
```shell
docker run --rm -it -p 8000:8000 -p 2303:2303 -p 1905:1905 -p 8765:8765 -p 6080:6080 -p 1108:1108 jderobot/robotics-academy:latest ./start-3.1.sh
```

I can raise the PR if it is required.

pariaspe, 2022-04-15:
Hi @Apoorvgarg-creator, thanks for noticing. Launching instruction at Laser Mapping exercise is quite out of date, we are currently at version 3.2.5.
Latest image (which points to 3.2.5) included some changes in the exercise launching. I'm updating launching instructions soon (#1643), but it will be great if you can try launching the exercise with the following command:
```
docker run --rm -it -p 8000:8000 -p 2303:2303 -p 1905:1905 -p 8765:8765 -p 6080:6080 -p 1108:1108 jderobot/robotics-academy
```

Apoorvgarg-creator, 2022-04-15:
@pariaspe I have seen whenever I hover over a card in Robot Academy exercise page, it shows a blank result. 

Screen recording -

https://user-images.githubusercontent.com/57873504/163565828-b7d81a51-2377-4142-a3f1-8f8d00455ce1.mov

pariaspe, 2022-04-15:
Not sure if it's the expected behavior or it may be a bug on css/html files. Feel free to open an issue related to it.
(Please try to not mix unrelated content to the same issue)

Apoorvgarg-creator, 2022-04-15:
> Not sure if it's the expected behavior or it may be a bug on css/html files. Feel free to open an issue related to it.
> (Please try to not mix unrelated content to the same issue)

Sorry for the mixup. 
I will open another issue regarding this.

pariaspe, 2022-05-03:
Done in #1692

PR #1692 (2022-05-03): Updated RADI Launch Instructions
### Updated RADI Launch Instructions according to the New Launch Instructions.

**Syntax Followed:-**

- Standard launch:

```
docker run -it --rm -p <ports> jderobot/robotics-academy
```

- GPU acceleration launch:

```
docker run -it --rm --device /dev/dri -p <ports> jderobot/robotics-academy
```

- Store terminal output launch:

```
docker run -it --rm -v $HOME/.roboticsacademy/log/:/root/.roboticsacademy/log/ -p <ports> jderobot/robotics-academy --logs
```

fixes #1643
fixes #1562
fixes #1669
fixes #1683
closed 2022-05-03 by pariaspe