JdeRobot/RoboticsAcademy #1670: System Crashes after Launching RobotAcademy

Blebot0, 2022-04-13:
### System Details:
- OS: Fedora 35
- RADI Version: Latest

## Installation Steps Followed:
`docker pull jderobot/robotics-academy:latest`

## Details of the issue:
After running the command:
`docker run --rm -it -p 8000:8000 -p 2303:2303 -p 1905:1905 -p 8765:8765 -p 6080:6080 -p 1108:1108 jderobot/robotics-academy:latest ./start.sh`

I navigate to localhost:8000 on my web browser. After the "Connection" was established, when I click on Launch button, my system stops responding.
Source : (https://jderobot.github.io/RoboticsAcademy/exercises/AutonomousCars/follow_line/)

I am unable to take screenshots of the output screen on my terminal.

pariaspe, 2022-04-15:
Hi @Blebot0 , please try again on latest docker image (3.2.5) with the following command:
```
docker run --rm -it -p 8000:8000 -p 2303:2303 -p 1905:1905 -p 8765:8765 -p 6080:6080 -p 1108:1108 jderobot/robotics-academy
```

Blebot0, 2022-04-16:
@pariaspe I tried it. Still the same issue.

pariaspe, 2022-04-18:
Could you share your console output?

Blebot0, 2022-04-18:
I'll try to log my terminal output. As I told you, my entire system freezes as soon as I click launch which is followed by a hard reset.

akshaynarisetti, 2022-04-25:
Hi @Blebot0 

I am trying to replicate this issue on my end, Is there any sort of information or steps you have followed during the launch which is not mentioned above.

akshaynarisetti, 2022-04-26:
@pariaspe I am able to replicate the issue on my end.

Here is the console output https://pastebin.com/0fkqX9Ud, something looks familiar?

pariaspe, 2022-04-27:
Hey @akshaynarisetti , try to run RADI with `--logs` flag. It will record logs into `.roboticsacademy/log/{date_time}`

akshaynarisetti, 2022-04-27:
Hi @pariaspe,

I tried using `--logs` flag, doesn't seem to be the correct way, It throws an error

```
unknown flag: --logs
See 'docker run --help'.
```

I tried with `-logs` flag which logged to `/var/lib/docker/containers/` but didn't find anything much there, here is the logs for your reference.

https://github.com/akshaynarisetti/issue1670_logs/tree/main/a46b603f60d6c50df76fbd002b96654e89e3622a06b8f411cc31f6b80af020b1

akshaynarisetti, 2022-04-27:
Although the std::bad_alloc() seems to be caused by not enough memory being left in the device, I still haven't come to a conclusion on the root cause.

I'll try other ROS docker containers and see if they are working fine on Fedora_35.

pariaspe, 2022-04-27:
> Hi @pariaspe,
>
> I tried using `--logs` flag, doesn't seem to be the correct way, It throws an error
>
> ```
> unknown flag: --logs
> See 'docker run --help'.
> ```
>
> I tried with `-logs` flag which logged to [...]

Try:
```
docker run --rm -it -p 8000:8000 -p 2303:2303 -p 1905:1905 -p 8765:8765 -p 6080:6080 -p 1108:1108 jderobot/robotics-academy --logs
```

Blebot0, 2022-04-27:
@akshaynarisetti How did you log the output from command line? Are you using a VM for Fedora 35?

akshaynarisetti, 2022-04-27:
Hi @Blebot0 I am running it on AWS cloud essentially  in virtualization env.

akshaynarisetti, 2022-04-28:
Hi @pariaspe,

logs doesn't seem to work.

Could be because docker crashes along with the terminal application.

akshaynarisetti, 2022-04-28:
Hi @pariaspe 

Update:- I tried running the official Docker image of ros:noetic. It also throws the same error on executing `roscore`.

Seems the issue is not with our Repository.

The core issue is with Fedora_35, Docker and ROS.

pariaspe, 2022-04-29:
Thanks @akshaynarisetti for testing on Fedora35 machines

akshaynarisetti, 2022-08-22:
Hi @pariaspe

Let us escalate this issue to Docker or Fedora and this issue should be closable once done.

javizqh, 2024-07-03:
This issue now longer occurs in Fedora 37 with the current instructions found in [here](https://github.com/JdeRobot/RoboticsAcademy) and the current RoboticsBackend version 4.6.1

closed 2024-07-03 by javizqh