JdeRobot/RoboticsAcademy #1671: Discussion about exercise containing 2 web template

Apoorvgarg-creator, 2022-04-14:
According to the [RepositoryArchitecture.md](https://github.com/JdeRobot/RoboticsAcademy/blob/master/docs/RepositoryArchitecture.md) exercise.html is present in `/exercises/templates/exercises` But for some exercise for ex. Color filter exercise  [here](https://github.com/JdeRobot/RoboticsAcademy/tree/master/exercises/static/exercises/color_filter/web-template/WebRtc/web-template) also have exercise.html. I tried understanding the difference so I ran both on local server, and the know that they are different in look but I wanted to know when and how it is used ?
Can I know what's the difference between these two exercise.html ?

pariaspe, 2022-04-15:
Hi @Apoorvgarg-creator, most of HTML files out of `/exercises/templates/exercises/` are unused and deprecated, so they are susceptible to be deleted. 
Thanks for noticing.

Apoorvgarg-creator, 2022-04-15:
Thank you @pariaspe for the help

jmplaza, 2022-08-15:
The confusion due to several web templates in some exercises will be solved in #1792

GauravKumar9920, 2023-01-28:
Is this issue solved ?

pariaspe, 2023-01-30:
Yes, it is solved. Thanks @GauravKumar9920

closed 2023-01-30 by pariaspe