JdeRobot/RoboticsAcademy #1675: [manager 4.0] error gazebo stats

Carlosalpha1, 2022-04-20:
Hi, I have a **problem** with [**manager-4.0.py**](https://github.com/JdeRobot/RoboticsAcademy/blob/foxy/scripts/manager-4.0.py) of foxy branch.  
When I launch my new exercise **Follow Person** the web-template crashes showing **Connecting 2...** indefinitely.

But, if I comment these lines, the **exercise** and **VNC** (with gazebo and terminal) work well.

[image]  

I have tried to find the threads running using **pstree** command:  

[image]

Any idea what may be happening?
I think the problem is that the exercise needs a lot of **Gazebo resources** and using **gz stats** blocks the launch

Cheers!

akshaynarisetti, 2022-04-28:
Hi @Carlosalpha1 
Did you try running that code in an individual thread?

Tejal-19, 2023-03-26:
Yes, I think as it is running a subprocess, it is running multiple secondary processes. I think running an individual thread can be an option.

jmplaza, 2023-03-26:
Manager-4.0.py was a nice prototype exploring ROS2 Foxy, but the new manager (=RAM, Robotics Application Manager) should be checked to know whether or not this error reporting Gazebo stats keeps bothering or not. That RAM will be used for ROS1-Noetic and for ROS2- Humble RADIs in the long run. Regarding ROS1-Noetic it is included in RADI-3.3.1 and available for the FollowLine exercise and the BasicVacuumCleaner exercise.

jmplaza, 2023-04-05:
AFAIK the new manager, named RAM, is working fine in RADI 4.3.X and properly receiving the gazebo stats (such as Real Time Factor, RTF), which uses ROS2 Humble as reference. 

@Blancasr is this right?. In such case we could close this issue as solved.

jmplaza, 2023-04-20:
Confirmed, it is working fine in ROS2 Humble RADI.

closed 2023-04-20 by jmplaza