JdeRobot/RoboticsAcademy #1683: ./start-3.1.sh: no such file or directory: unknown.

ibrahimgb, 2022-04-25:
Docker version 20.10.7, build 20.10.7-0 ubuntu 5~20.04.2

I am getting this ERROR: 
docker: Error response from daemon: failed to create shim: OCI runtime create failed: container_linux.go:380: starting container process caused: exec: "./start-3.1.sh": stat ./start-3.1.sh: no such file or directory: unknown.

I followed this guide to install it : [Laser Mapping](https://jderobot.github.io/RoboticsAcademy/exercises/MobileRobots/laser_mapping)

this issue i head after running this code:
``` docker run --rm -it -p 8000:8000 -p 2303:2303 -p 1905:1905 -p 8765:8765 -p 6080:6080 -p 1108:1108 jderobot/robotics-academy:3.1.4 ./start-3.1.sh ```

it's my first time here so I don't know if the problem is in my Docker version or something else

pariaspe, 2022-04-25:
Hi @ibrahimgb,

Try the following command:
```
docker run --rm -it -p 8000:8000 -p 2303:2303 -p 1905:1905 -p 8765:8765 -p 6080:6080 -p 1108:1108 jderobot/robotics-academy
```

akshaynarisetti, 2022-04-25:
Hello, @ibrahimgb

I am trying to replicate this issue on my end, Is there any sort of information or steps you have followed during the launch which is not mentioned above.

pariaspe, 2022-04-25:
This issue is already tracked in #1643. Instructions launching are deprecated with RADI 3.2.5 and have to be updated as the issue indicates

ibrahimgb, 2022-04-25:
thanks, @pariaspe that works for me!
so the installation instruction needs to be updated.

jmplaza, 2022-04-26:
Let's wait until the PR with the updated installation instructions is merged to close this issue.

akshaynarisetti, 2022-04-26:
Hi @jmplaza 
Might want to tag this issue with gh-pages Label 🏷

pariaspe, 2022-05-03:
Done in #1692

PR #1692 (2022-05-03): Updated RADI Launch Instructions
### Updated RADI Launch Instructions according to the New Launch Instructions.

**Syntax Followed:-**

- Standard launch:

```
docker run -it --rm -p <ports> jderobot/robotics-academy
```

- GPU acceleration launch:

```
docker run -it --rm --device /dev/dri -p <ports> jderobot/robotics-academy
```

- Store terminal output launch:

```
docker run -it --rm -v $HOME/.roboticsacademy/log/:/root/.roboticsacademy/log/ -p <ports> jderobot/robotics-academy --logs
```

fixes #1643
fixes #1562
fixes #1669
fixes #1683
closed 2022-05-03 by pariaspe