JdeRobot/RoboticsAcademy #1693: RoboticsAcademy on goodfirstissue.dev

pariaspe, 2022-05-03:
### Discussed in https://github.com/JdeRobot/RoboticsAcademy/discussions/1690

<div type='discussions-op-text'>

<sup>Originally posted by **akshaynarisetti** April 29, 2022</sup>
Hello Everybody 👋,

Should we put RoboticsAcademy Repo on the https://goodfirstissue.dev/ site, it’s basically where budding open source contributors would go to and search on what open source projects they should work on.

Putting our Repo on such sites can increase the visibility to our project.</div>

Nice idea @akshaynarisetti

akshaynarisetti, 2022-05-03:
Hi @pariaspe @jmplaza 
I don’t have required permissions to add labels to issues which is required to add this project to https://goodfirstissue.dev/ as given in the instructions at https://github.com/deepsourcelabs/good-first-issue#adding-a-new-project

pariaspe, 2022-05-03:
Hey @akshaynarisetti , I'm on it! Also, we have to work on a CONTRIBUTING.md

akshaynarisetti, 2022-05-03:
Meanwhile, I will be working on the README.md, let’s make our README Look out of the world!

pariaspe, 2022-05-04:
https://github.com/deepsourcelabs/good-first-issue/pull/413

akshaynarisetti, 2022-05-05:
Hi @pariaspe 
Do I get a free hand in writing or redefining RoboticsAcademy Tagline and other things in the README, I'll obviously get it vetted by everyone once I author out the README.

pariaspe, 2022-05-05:
You can share your suggestions first or open a PR and we can discuss the changes

pariaspe, 2022-05-13:
Merged

closed 2022-05-13 by pariaspe