JdeRobot/RoboticsAcademy #1815: Refactoring of the manager with a new protocol

dvalladaresv, 2022-09-16:


dmariaa, 2022-09-19:
Linked a new branch, "new-architecture" to work on this issue...

More info about it in this discussion #1799

jmplaza, 2022-12-26:
@dmariaa  has updated the "new-architecture" branch to the latest achievements. The RAM there now supports  "level 1 messages" (Gazebo launching and exercise.py launching), and "level 2 messages" (those about brain loading, execution, GUI messages). 

This RAM implements a wrapping for the code-websocket and gui-websocket from exercise.py. So the exercise.py still opens those two websockets, but now they are not directly connected to the web browser, where the exercise frontend is running. Instead, they are now connected to the new RAM, which redirects all the messages to/from them coming from/going to the browser.

@pariaspe @Apoorvgarg-creator please feel free to test it.

The problem with images at the FollowLine exercise is still there. We may overcome it using this RAM with Basic VacuumCleaner exercise instead, as it has no camera involved.