JdeRobot/RoboticsAcademy #1828: Introduce a search box in the starting web page

jmplaza, 2022-10-05:
So the users may search for a specific exercise by name, or even by tag. Maybe we may add some tags to attach to the exercise, to show its topic: autonomous driving, drones, mobile robots, service robots, computer vision, industrial robotics....

Once something is written in the search box, only the exercise matching it are displayed in the exercise grid.

Apoorvgarg-creator, 2022-10-09:
@dmariaa Can I take the role of working on the search and tag functionality ?

dmariaa, 2022-10-09:
Absolutly Yes @Apoorvgarg-creator .... I would consider adding a field to the exercise model in the Django backend, something like "tags"... then you can use that field in the backend and frontend like we do with the rest... Let me know if you need any help...

Thanks!!!!

jmplaza, 2022-10-09:
Perfect. For the tags field in the exercise DabaBase we may have at least these tag names as possible values: Autonomous Driving, Service Robots, Drones, Computer Vision, Industrial Robots and DeepLearning, Mobile Robots. They cover all the currently existing exercises.

They are not disjoint. Each exercise may have attached one or more than one of them

dmariaa, 2022-10-09:
BTW... Tags have to be maintained (added, modified, removed) in the /admin section of robotics academy, the list frontend need to have the possibility of being filtered by this tag, or searched by name...

PR #1838 (2022-10-22): New: Search functionality and New Homepage
Fixes #1828

This PR do the following tasks to fix the above the issue -
1. New Exercise Card Style following material design guidelines
2. Search Bar in the Main App bar
3. Search filters only on the basis of exercise name
4. Introduction of tags in the exercises database
5. Introduction of status in the exercise database
6. Status and tags can be change via Admin login in Django
7. Status currently offers only three choice - "active", "inactive" and "prototype".

Future Work -
1. Tags for every exercises needs to be populated
2. Status needs to be f [... cut]
closed 2022-10-22 by dvalladaresv