JdeRobot/RoboticsAcademy #1849: Error while running Instructions for developers. Webpack-stats.json doesn't exist.

PabloAsensio, 2022-11-06:
I've just clone the repo, I created a new local branch and I tried to follow [Instructions for developers](https://github.com/JdeRobot/RoboticsAcademy/blob/master/docs/InstructionsForDevelopers.md)

I'm using Apple M2 chip.

**ERROR**
[image]

**DOCKER**
[image]

**ENV PYTHON PACKGES**
[image]

What I need to do to run the http://127.0.0.1:8000/ properly?

dvalladaresv, 2022-11-09:
You have to install the dependencies for react   (with Yarn or npm, required Node.JS >= 14.16)  
```
cd react_frontend/ && yarn install && yarn run dev    
```   
The documentation has just been improved by adding this part

closed 2022-11-09 by dvalladaresv

PabloAsensio, 2022-11-09:
Thanks. Only with ```cd react_frontend/ && yarn install ```  is fine. Also next step is usung python key word, not python3. We are using virtualenv