JdeRobot/RoboticsAcademy #1890: [RADI] Why we don't use more symbolic links?

PabloAsensio, 2022-12-04:
It tookme 2 days to fix a issue I had.

I was trying to click on the ```launch``` button of the exercice I am doing, but when I updated the [```instructions.json```](https://github.com/JdeRobot/RoboticsAcademy/blob/master/scripts/instructions.json) nothing works. I though that I made a mistake becase my ```exercise``` var in ```exercise.html``` apparently was broken. I had to overwrite the ```instructions.json``` in / because it is not a symbolic link.

It is because dakefile copy the  ```instructions.json``` in ```/``` but it sould be a link to this.
**Please, change line 49 in Docker file to symbolic links.**

```Dockerfile
COPY [".env", "manager.py", "instructions.json", "pylint_checker.py", "entrypoint.sh", "./"]
````

Maybe should be interesting to do the same with line 42
```Dockerfile
RUN rsync -a --exclude 'ace-builds' /RoboticsAcademy/exercises/static/exercises/* /RoboticsAcademy/exercises
```

pariaspe, 2022-12-05:
Hi @PabloAsensio, I don't understand your suggestion. Where does the link should point to?

PabloAsensio, 2022-12-05:
Hi @pariaspe,

Basicly I mean this. See ```instructions.json```  and ```manager.py``` 

[image]

pariaspe, 2022-12-05:
I guess you get that error because you were modifying the file (the one from /RoboticsAcademy/...) from inside the container.  

If you modify it locally and then build the image, you won't face that problem.

Anyway, let's explore the use of symlinks instead :+1

closed 2024-10-30 by javizqh