JdeRobot/RoboticsAcademy #1900: [follow_line] Remove world duplicities

pariaspe, 2022-12-30:
F1 models are now included inside the world file, which implies having two world files (and also two launch files) for each circuit (classic and game). The only difference between them is having a second f1 model (`f1_guest.sdf`) in game worlds.
```xml
<?xml version="1.0" ?>
<sdf version="1.5">
  <world name="default">
    <gui fullscreen=1></gui>
    <scene>
      <grid>false</grid>
      <sky>
        <clouds>
          <speed>12</speed>
        </clouds>
      </sky>
    </scene>
    <!-- A global light source -->
    <include>
      <uri>model://sun</uri>
      <name>sun_1</name>
      <pose>0 0 1 0 0 0</pose>
    </include>
    <include>
	    <uri>model://simple_circuit</uri>
	    <pose>0 0 0 0 0 0</pose>
    </include>
    <include>
      <uri>model://f1</uri>
      <pose>53.462 -10.734 0.004 0 0 -1.57</pose>
    </include>
    <include>
      <uri>model://f1_guest</uri>
      <pose>-53.2 -6.8 0.004 0 0 -4.85</pose>
    </include>
  </world>
</sdf>
```

World file could be the same for both launch files if the second f1 (f1_guest) is spawn directly from game launch files. Model spawn could be done with:
```xml
<!-- Spawn a robot into Gazebo -->
<node name="spawn_urdf" pkg="gazebo_ros" type="spawn_model" args="-file $(find baxter_description)/urdf/baxter.urdf -urdf -z 1 -model baxter" />
```

shree-10, 2023-01-11:
i want to work on this issue will u please assign me this issue and guide me ...

pariaspe, 2023-01-11:
Hi @shree-10 , thanks for your interest. I think the issue is well self-explained, but what do you need guide for? 
Check our [contributing guide](https://github.com/JdeRobot/RoboticsAcademy/blob/master/CONTRIBUTING.md) to learn how to help us. Basically, you need to fork the repository and open a PR with your proposed solution. After that, I'll check it.

GauravKumar9920, 2023-01-27:
Hey,
I have worked on the issue and have a possible solution to it where shall I send the updated code for review ?

jmplaza, 2023-01-27:
Just create a PullRequest (PR) containing your proposed changes from the master branch mentioning the issue. Then a committer will review you PR

GauravKumar9920, 2023-01-27:
I have created a PR kindly go through it.

MashiatK, 2023-02-08:
I want to work on this. If the issue is open, please assign it to me.

jmplaza, 2023-02-08:
Hi @MashiatK , there is no need for explicit assignment, just work on it and feel free to submit a Pull Request solving it :-). Also take into account that @GauravKumar9920 has already submitted a PR

pariaspe, 2023-02-15:
Solved in #1909

closed 2023-02-15 by pariaspe