JdeRobot/RoboticsAcademy #1910: [follow_line] Model visual details not displaying well

pariaspe, 2023-01-30:
F1 model shows up like:
[image]

while should be like:
[image]

NAFI5O, 2023-01-31:
Hello @pariaspe, I want to contribute to this issue. How can I go further?

NAFI5O, 2023-02-01:
Hello @pariaspe, Thank for guiding me,Can you tell me how I can learn the full system overview of this repo,I even don’t understand how thinks work and how I can contribute to this repo.I will be grateful if you give me your email address.

Thanks in advance.

pariaspe, 2023-02-01:
> Hello @pariaspe, Thank for guiding me,Can you tell me how I can learn the full system overview of this repo,I even don’t understand how thinks work and how I can contribute to this repo.I will be grateful if you give me your email address.
> 
> Thanks in advance.

I'm sorry but I don't use to share my email address. The best way for learning is by doing. Have you pulled the docker image and tried RoboticsAcademy? You can find the full system overview in the [docs](https://github.com/JdeRobot/RoboticsAcademy/tree/master/docs) folder.

NAFI5O, 2023-02-01:
@pariaspe What I have done by now that fork and clone the repo to my ubuntu system. So how I can go further from here? I have installed docker, Django, and other dependencies.

pariaspe, 2023-02-01:
Start trying one of the exercises, by instance [follow_line](https://jderobot.github.io/RoboticsAcademy/exercises/AutonomousCars/follow_line/). After that, try to build your own RADI, instead of using the official one (that you must have pulled before). Make some modifications to change the f1 model (as I explained before) and test them in a new built RADI. You may want to test it locally (not using the docker image) before.

GauravKumar9920, 2023-02-01:
Hey, from what I understood in this issue we just need to spawn a different model  (f1_renault) instead of what is already given (f1) in the classical world right (noting that a different model (f1_renault) would spawn in the game world too ) am I right ?

pariaspe, 2023-02-01:
> Hey, from what I understood in this issue we just need to spawn a different model (f1_renault) instead of what is already given (f1) in the classical world right (noting that a different model (f1_renault) would spawn in the game world too ) am I right ?

Yes, that's right. Also, some testing is needed to check that everything work :+1: .

GauravKumar9920, 2023-02-02:
For this Issue do i also need to edit the manager.py files to reflect the changes, cause when I changed it in the world file the changes were't reflecting

GauravKumar9920, 2023-02-02:
Hey, I feel I am launching the RAID it wrong configuration, I'm changing the models in the world but while running the RADI I'm getting the same model in gazebo as in the orignal repo, can someone guide me through the issue..
I'm running this command for running the RADI - docker run --rm -it -p 8000:8000 -p 2303:2303 -p 1905:1905 -p 8765:8765 -p 6080:6080 -p 1108:1108 -v $(pwd):/home jderobot/robotics-academy

pariaspe, 2023-02-03:
> Hey, I feel I am launching the RAID it wrong configuration, I'm changing the models in the world but while running the RADI I'm getting the same model in gazebo as in the orignal repo, can someone [...]

Hey, it seems that your are not running the docker image you want to. What I normally do is building the RADI (manually or with [build.sh](https://github.com/JdeRobot/RoboticsAcademy/blob/master/scripts/build.sh) script) with some test like `issue-1910` or `test`. After that, open a bash in that RADI (`docker run -it jderobot/robotics-academy:<tag> --debug`, change the `<tag>` with actual tag you used when building the RADI) and check manually if that is the one you want to run (by instance, check that the files your modifying have the changes you did). When, you are sure of that, run it as always `docker run --rm -it -p 8000:8000 -p 2303:2303 -p 1905:1905 -p 8765:8765 -p 6080:6080 -p 1108:1108 jderobot/robotics-academy:<tag>`.
PS: Why are you mounting a volume in the command you pasted? I may be skipping something, but I think you don't need that for testing it.

closed 2023-02-15 by pariaspe