JdeRobot/RoboticsAcademy #1914: [follow_line] Incorrect initial gazebo view

pariaspe, 2023-02-03:
Change the initial gazebo view position to a better one. It can be done by adding these lines to the existing worlds:
```xml
<gui fullscreen='0'>
    <camera name='user_camera'>
        <pose frame=''>x y z R P Y</pose>
        <view_controller>orbit</view_controller>
        <projection_type>perspective</projection_type>
    </camera>
</gui>
```

sameergupta4873, 2023-02-03:
@pariaspe I think this is pretty straight forward, can I work on this ?

MashiatK, 2023-02-08:
I want to work on this. If the issue is open, please assign it to me.

jmplaza, 2023-02-08:
Hi @MashiatK , there is no need for explicit assignment, just work on it and feel free to submit a Pull Request solving it :-)

GauravKumar9920, 2023-02-18:
Hey @pariaspe, I am updating the initial view location of the world with following views -
for Default line - 
[image]
for montmelo -
[image]
for montreal - 
[image]
and the last one N..-
[image]

if you suggest any changes do lemme know.

pariaspe, 2023-02-18:
Hi, all of them look fine to me!

pariaspe, 2023-02-18:
Solved in #1936 , thanks @GauravKumar9920 !

closed 2023-02-18 by pariaspe