JdeRobot/RoboticsAcademy #1915: [vacuum_cleaner] Incorrect initial gazebo view

pariaspe, 2023-02-03:
Change the initial gazebo view position to a better one. It can be done by adding these lines to the existing worlds:
```xml
<gui fullscreen='0'>
    <camera name='user_camera'>
        <pose frame=''>x y z R P Y</pose>
        <view_controller>orbit</view_controller>
        <projection_type>perspective</projection_type>
    </camera>
</gui>
```

GauravKumar9920, 2023-02-04:
Hey, I would like to work on this, I will send a pr after I test it...

AkashS20, 2023-02-04:
Hi, I would like to contribute to this. I'm new to open source so if you could please guide me on solving this,
it will be of great help.

pariaspe, 2023-02-04:
Open a PR proposing a solution and I'll review it when I have time. You may find useful our [contribution guide](https://github.com/JdeRobot/RoboticsAcademy/blob/master/CONTRIBUTING.md).

AkashS20, 2023-02-06:
@pariaspe could you elaborate on what `x y z R P Y` should be replaced by? I think it should be replaced by _desired position and orientation of the view in the world,_ but I am unable to understand what desired position and orientation is.

jmplaza, 2023-02-06:
Hi @AkashS20 ,

no, that is part of the task :-). Use the Basic Vacuum Cleaner exercise on RoboticsAcademy and check the initial position of the camera. That one is not convenient at all, a better camera position would be a bird's eye view. From there the whole apartment and the vacuum cleaner can be seen at a glance. Once you use that exercise you will understand it. 

Cheers,

dsg1320, 2023-02-16:
Hi, I would like to work on this issue. Could you guide me through solving this since I'm a beginner?

pariaspe, 2023-02-17:
Hi @dsg1320 , yes please, go ahead and open a PR with your solution. You may find useful our [contribution guide](https://github.com/JdeRobot/RoboticsAcademy/blob/master/CONTRIBUTING.md). The issue is pretty straightforward, but this [link](https://answers.gazebosim.org//question/408/is-there-a-way-to-modify-the-default-camera-position-in-the-world-file/) may help too.

GauravKumar9920, 2023-02-18:
Is there any preferred view angle ?
cause there's a lot of angles form which I can look at a bot...

GauravKumar9920, 2023-02-18:
what's the preferred one ?
view -1
[image]
oe view -2
[image]

pariaspe, 2023-02-18:
Hey! What about a top view where it can be seen all the house? Thanks!

GauravKumar9920, 2023-02-18:
okkayy, I will do it thanks :)

pariaspe, 2023-02-20:
Solved in #1937 , thanks @GauravKumar9920 !

closed 2023-02-20 by pariaspe