JdeRobot/RoboticsAcademy #1921: Implement GPU acceleration in RAM

ango1994, 2023-02-06:
Enable GPU acceleration in the new Robotics Application Manager

jmplaza, 2023-02-06:
Related to https://github.com/JdeRobot/RoboticsAcademy/issues/902. @dpascualhe is also working on refining the support of GPU acceleration both on Linux user machines and on Windows user machines.

Bhardwaj-Himanshu, 2023-03-25:
Hi **@ango1994 @jmplaza**, (sorry for the ping, but do let me know if it bothers you)
I want to start contributing to this issue, just wanted to confirm a few steps related to it.

-after setting the codebase up locally, which files do I need to navigate into to implement the same?
-also, if you could provide a slight idea about how to start?

--**Some various methods after my little walkthrough suggested usage of CUDA or pytorch!**

I know these doubts can sound naive, but yep these do derive from my lack of experience!
But I would love to know about this, cause I am very keen into "how we implement hardware usage and can control it, via our code."

ango1994, 2023-03-28:
HI @Bhardwaj-Himanshu check the RAM (Robotics Application Manager).
https://github.com/JdeRobot/RoboticsAcademy/blob/master/src/manager/manager.py 
It's a new implementation of the manager more agnostic to the aplication that is launched. The RAM should implement GPU acceleration as the old manager. https://github.com/JdeRobot/RoboticsAcademy/blob/master/scripts/manager.py

Bhardwaj-Himanshu, 2023-04-08:
Hi @ango1994 ,
If this issue is being still worked on, I would like to state my thinking--

https://github.com/JdeRobot/RoboticsAcademy/blob/b918ce1a73063b0b3aa9388371e33151630dec41/src/manager/manager.py#L195-L199

The highlighted lines in above codebase are from new manager.py script

So,should I try to implement the steps mentioned here below-
https://github.com/JdeRobot/RoboticsAcademy/blob/b918ce1a73063b0b3aa9388371e33151630dec41/scripts/manager.py#L66-L82

Do let me know, would be of great help.|
Also does ROS in the previous script same as RAM in new one?