JdeRobot/RoboticsAcademy #1944: New AutoParking betatesting

pariaspe, 2023-02-21:


GauravKumar9920, 2023-02-22:
How do I set it up manually ?
I can Find one Autoparking exercise in the docker.
can you provide some more details

GauravKumar9920, 2023-02-23:
In beta testing I can setup the Gazebo world and spawn, but the web server and docker side of the part how do I build it @pariaspe

GauravKumar9920, 2023-02-23:
Also in the current autoparking excersise which is present in the RADI the world and the models aren't spawning.

pariaspe, 2023-02-23:
Hi, follow the existing [instructions](https://github.com/JdeRobot/RoboticsAcademy/tree/master/scripts#readme). Everything is supposed to spawn inside the RADI. If models are not spawning in gazebo your environment variables may not be set (or wrong set). Check in **GAZEBO_MODEL_PATH** and **GAZEBO_RESOURCE_PATH** variables if you can see models and worlds paths.

GauravKumar9920, 2023-02-24:
hey @pariaspe , actually I spotted a bug in the [Autoparking](https://github.com/JdeRobot/RoboticsAcademy/tree/master/exercises/static/exercises/autoparking) exercise, the models weren't spawnign in that I have fixed it. shall I create an issue for the same as the issues is persistent in the og repo ?
Also, just to be sure are we supposed to be working with [autoparking_v2](https://github.com/JdeRobot/RoboticsAcademy/tree/master/exercises/static/exercises/autoparking_v2) If yes can you guide me to build the same so that it reflects in the locally build docker image as an excersise.

GauravKumar9920, 2023-02-26:
Hey @pariaspe any comments 😅

pariaspe, 2023-02-26:
Hey @GauravKumar9920 , please avoid to ping me twice, I'll answer your questions asap.

I don't spot any bug, since the models are spawning correctly. Anyway, what is your suggestion?

[image]

You need to add autoparking_v2 exercise to the exercises database. It is explained in the [documentation](https://github.com/JdeRobot/RoboticsAcademy/blob/master/docs/InstructionsForDevelopers.md#how-to-add-a-new-exercise). You need to build a new RADI first, because the latest doesn't contain the exercise

GauravKumar9920, 2023-02-27:
Okkay I will make sure of that.
Actually I checked the following by running the latest RADI, it doesn't has that problem of not being able to spawn the bot.
but for the same the code in the latest RADI was different 
the code I found on the RoboticsAcademy/scripts/.env by running the latest RADI was -
`#!/bin/bash

# GAZEBO
export GAZEBO_RESOURCE_PATH=$GAZEBO_RESOURCE_PATH:/usr/share/gazebo-11:/RoboticsAcademy/exercises/follow_line/web-template/launch/:/RoboticsAcademy/exercises/obstacle_avoidance/web-template/launch/:/RoboticsAcademy/exercises/vacuum_cleaner/web-template/launch/:/RoboticsAcademy/exercises/vacuum_cleaner_loc/web-template/launch/:/RoboticsAcademy/exercises/3d_reconstruction/web-template/launch/:/RoboticsAcademy/exercises/global_navigation/web-template/launch/:/RoboticsAcademy/exercises/car_junction/web-template/:/RoboticsAcademy/exercises/opticalflow_teleop/web-template/launch/:/RoboticsAcademy/exercises/autoparking/web-template/:/RoboticsAcademy/exercises/montecarlo_visual_loc/web-template/launch/
export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:/PX4-Autopilot/build/px4_sitl_default/build_gazebo:/catkin_ws/devel/lib
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/usr/share/gazebo-11/models:/opt/jderobot/CustomRobots/f1/models:/opt/jderobot/CustomRobots/roomba_robot/models:/opt/jderobot/CustomRobots/3d_reconstruction/models:/opt/jderobot/CustomRobots/Taxi_navigator/models:/opt/jderobot/CustomRobots/car_junction/models:/PX4-Autopilot/Tools/sitl_gazebo/models:/catkin_ws/src/drones/drone_assets/models:/opt/jderobot/CustomRobots/package_delivery/models:/opt/jderobot/CustomRobots/power_tower_inspection/models:/catkin_ws/src/CrazyS/rotors_gazebo/models
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:/PX4-Autopilot/build/px4_sitl_default/build_gazebo
`

whereas in the [Repo](https://github.com/JdeRobot/RoboticsAcademy/blob/master/scripts/.env) it was - 
`#!/bin/bash

# GAZEBO

export GAZEBO_RESOURCE_PATH=$GAZEBO_RESOURCE_PATH:/usr/share/gazebo-11
export GAZEBO_RESOURCE_PATH=$GAZEBO_RESOURCE_PATH:$EXERCISES_STATIC_FILES/follow_line/web-template/launch/
export GAZEBO_RESOURCE_PATH=$GAZEBO_RESOURCE_PATH:$EXERCISES_STATIC_FILES/obstacle_avoidance/web-template/launch/
export GAZEBO_RESOURCE_PATH=$GAZEBO_RESOURCE_PATH:$EXERCISES_STATIC_FILES/vacuum_cleaner/web-template/launch/
export GAZEBO_RESOURCE_PATH=$GAZEBO_RESOURCE_PATH:$EXERCISES_STATIC_FILES/vacuum_cleaner_loc/web-template/launch/
export GAZEBO_RESOURCE_PATH=$GAZEBO_RESOURCE_PATH:$EXERCISES_STATIC_FILES/3d_reconstruction/web-template/launch/
export GAZEBO_RESOURCE_PATH=$GAZEBO_RESOURCE_PATH:$EXERCISES_STATIC_FILES/global_navigation/web-template/launch/
export GAZEBO_RESOURCE_PATH=$GAZEBO_RESOURCE_PATH:$EXERCISES_STATIC_FILES/car_junction/web-template/
export GAZEBO_RESOURCE_PATH=$GAZEBO_RESOURCE_PATH:$EXERCISES_STATIC_FILES/opticalflow_teleop/web-template/launch/
export GAZEBO_RESOURCE_PATH=$GAZEBO_RESOURCE_PATH:$EXERCISES_STATIC_FILES/autoparking/web-template/
export GAZEBO_RESOURCE_PATH=$GAZEBO_RESOURCE_PATH:$EXERCISES_STATIC_FILES/montecarlo_visual_loc/web-template/launch/
export GAZEBO_RESOURCE_PATH=$GAZEBO_RESOURCE_PATH:/usr/share/gazebo-11:/RoboticsAcademy/exercises/follow_line/web-template/launch/:/RoboticsAcademy/exercises/obstacle_avoidance/web-template/launch/:/RoboticsAcademy/exercises/vacuum_cleaner/web-template/launch/:/RoboticsAcademy/exercises/vacuum_cleaner_loc/web-template/launch/:/RoboticsAcademy/exercises/3d_reconstruction/web-template/launch/:/RoboticsAcademy/exercises/global_navigation/web-template/launch/:/RoboticsAcademy/exercises/car_junction/web-template/:/RoboticsAcademy/exercises/opticalflow_teleop/web-template/launch/:/RoboticsAcademy/exercises/autoparking/web-template/:/RoboticsAcademy/exercises/autoparking_v2/web-template/:/RoboticsAcademy/exercises/montecarlo_visual_loc/web-template/launch/
export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:/PX4-Autopilot/build/px4_sitl_default/build_gazebo:/catkin_ws/devel/lib

export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/usr/share/gazebo-11/models:
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/catkin_ws/src/CrazyS/rotors_gazebo/models
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/catkin_ws/src/drones/drone_assets/models
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/PX4-Autopilot/Tools/sitl_gazebo/models
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/jderobot/CustomRobots/f1/models
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/jderobot/CustomRobots/roomba_robot/models
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/jderobot/CustomRobots/3d_reconstruction/models
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/jderobot/CustomRobots/Taxi_navigator/models
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/jderobot/CustomRobots/car_junction/models
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/jderobot/CustomRobots/package_delivery/models
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/jderobot/CustomRobots/power_tower_inspection/models

`

can you guide me to what file the RADI latest is using, cause in the [docker-file](https://github.com/JdeRobot/RoboticsAcademy/blob/master/scripts/Dockerfile) the branch specified is master

GauravKumar9920, 2023-02-27:
Okkay I will make sure of that.

Actually I checked the following by running the latest RADI, it doesn't has that problem that I mentioned above of not being able to spawn the bot into the arena.
but for the same the code in the latest RADI was different 
the code I found on the RoboticsAcademy/scripts/.env by running the latest RADI was -

`#!/bin/bash
export GAZEBO_RESOURCE_PATH=$GAZEBO_RESOURCE_PATH:/usr/share/gazebo-11:/RoboticsAcademy/exercises/follow_line/web-template/launch/:/RoboticsAcademy/exercises/obstacle_avoidance/web-template/launch/:/RoboticsAcademy/exercises/vacuum_cleaner/web-template/launch/:/RoboticsAcademy/exercises/vacuum_cleaner_loc/web-template/launch/:/RoboticsAcademy/exercises/3d_reconstruction/web-template/launch/:/RoboticsAcademy/exercises/global_navigation/web-template/launch/:/RoboticsAcademy/exercises/car_junction/web-template/:/RoboticsAcademy/exercises/opticalflow_teleop/web-template/launch/:/RoboticsAcademy/exercises/autoparking/web-template/:/RoboticsAcademy/exercises/montecarlo_visual_loc/web-template/launch/`

whereas it's diffrent in the [Repo](https://github.com/JdeRobot/RoboticsAcademy/blob/master/scripts/.env).
can you guide me to _**what branch the RADI:latest is using**_, cause in the [docker-file](https://github.com/JdeRobot/RoboticsAcademy/blob/62451d22536ebcb22eee3a23bc2b633313493192/scripts/Dockerfile#L53) the branch specified is master.

Also, the **_issue_** I mentioned previously **_still occurs_** when I build the docker image locally using ./build.sh
So for that, the **_changes I made_** to the same for it to work was correcting the path by adding /launch after /web-template in the [line](https://github.com/JdeRobot/RoboticsAcademy/blob/62451d22536ebcb22eee3a23bc2b633313493192/scripts/.env#L17).

I have demonstrated it [here](https://github.com/GauravKumar9920/RoboticsAcademy/blob/8d32463701aa9643039d0bda1b5b4d3ed1b73152/scripts/.env#L17)....

I am a bit confused so kindly help me out with it 
Anyways I am working to add the autoparking_v2 excersise to the RADI locally after testing I will update you of the same.

PS: I guess the topic has diverted to the autoparking instead of autoparking_v2 exercise so Shall I create a new discussion or shall we continue it here..

pariaspe, 2023-02-28:
Hey, you are completely right. `/launch` is missing after `/web-template`. Thanks for noticing. Could you open a patch solving it?

RADI:latest should always use master branch.

PS: my bad, I'm renaming this issue :+1:

pariaspe, 2023-02-28:
Because master branch has evolved since we released that RADI.
I've just uploaded a new RADIv3.3.0 (also new latest). You can work from that stating point and go on

GauravKumar9920, 2023-03-01:
Hey I have added the exercise, yet I am facing a bit of issues can you help me out with them...
[image]

[image]

ango1994, 2023-03-02:
HI @GauravKumar9920. The steps to create a new exercise are the following:

1. Add the exercise to the database wich you already have done.
2. Create a folder in exercises/static/exercises with the exercise inside. The name of the folder has to be the same of the exercise id you gave in the database.
3. Create a folder in exercises/templates/exercises with a exercise.html inside. The name of the folder also has to be the same of the exercise id.

GauravKumar9920, 2023-03-04:
I have cross-checked all the above mentioned points but the errors still occur.
These are all the [files](https://github.com/GauravKumar9920/RoboticsAcademy/pull/8/files) I have changed(compared to the master branch) you can have a look at them..
Also, This is the testing [branch](https://github.com/GauravKumar9920/RoboticsAcademy/tree/1944-testing) Where I am testing all the changes I made.
The errors that occurred - 
(one is after opening the exercise )-
[image]
one after running (`python3.8 manage.py runserver`) -
[image]

also the files mentioned in the above response -
1. done
2. [check 2](https://github.com/GauravKumar9920/RoboticsAcademy/tree/1944-testing/exercises/static/exercises/autoparking_v2)
3. [check 3](https://github.com/GauravKumar9920/RoboticsAcademy/tree/1944-testing/exercises/templates/exercises/autoparking_v2)

pariaspe, 2023-03-04:
Hi @GauravKumar9920 , I've just tested to see what's happening. I faced the same problem. How are adding the exercise to the database? By default, configuration is set as a string while it should be a json. Try, to replace `"{}"` with `{}`. 

[image]

I've also found another bug in `instruction.json` file. Paths for Autoparking_v2 exercise are wrong, they are skipping `static/exercises/` subdirectories. Compare them with old autoparking paths and you will notice easily.

```json
    "autoparking": {
        "gazebo_path": "/RoboticsAcademy/exercises/static/exercises/autoparking/web-template/launch",
        "instructions_ros": ["/opt/ros/noetic/bin/roslaunch ./RoboticsAcademy/exercises/static/exercises/autoparking/web-template/launch/autoparking.launch"],
        "instructions_host": "python3 /RoboticsAcademy/exercises/static/exercises/autoparking/web-template/exercise.py 0.0.0.0"
    },
    "autoparking_v2": {
        "gazebo_path": "/RoboticsAcademy/exercises/autoparking_v2/web-template/launch",
        "instructions_ros": ["/opt/ros/noetic/bin/roslaunch ./RoboticsAcademy/exercises/autoparking_v2/web-template/launch/autoparking.launch"],
        "instructions_host": "python3 /RoboticsAcademy/exercises/autoparking_v2/web-template/exercise.py 0.0.0.0"
    },
```

GauravKumar9920, 2023-03-04:
Thanks @pariaspe !
I will fix the path. 
also I will try replacing the same )

pariaspe, 2023-03-06:
Hey, I've fixed that on #1962 , so if you build RADI from master again, and you add the exercise as always it should work,

GauravKumar9920, 2023-03-07:
> Hey, I've fixed that on #1962 , so if you build RADI from master again, and you add the exercise as always it should work,

I checked the same twice once by implementing the changes onto the branch I was testing changes on, and second by[ cloning](https://github.com/JdeRobot/RoboticsAcademy/tree/1961-bad-configuration-default-in-exercise-model) the branch where the changes were made
It isn't working :,)

pariaspe, 2023-03-08:
It should work. @ango1994 tested this morning and merged the branch. Could you try again?

GauravKumar9920, 2023-03-09:
Hey @pariaspe I have sucessuflly added the exercise and made the required changes for it to run.
However we I have spotted a bug in the exercise- 
[image]
The world isn't spawning correctly, shall we work out over here or a new issue ?

pariaspe, 2023-03-09:
Nice, good work! Let's keep talking here since we are debugging it. Have you checked the GAZEBO_MODEL_PATH env var?

GauravKumar9920, 2023-03-09:
Hey, thanks !
I am working on it.

GauravKumar9920, 2023-03-18:
hey @pariaspe I am getting this error after running -
docker run --rm -it -p 8000:8000 -p 2303:2303 -p 1905:1905 -p 8765:8765 -p 6080:6080 -p 1108:1108 jderobot/robotics-academy:testing

[image]

ango1994, 2023-03-18:
Hi @GauravKumar9920 you need to rebuild the base image since there have been changes in the dependencies. "./build.sh -f test"

GauravKumar9920, 2023-03-18:
kindly check the PR, I have fixed the world not being able to spawn issue.

pariaspe, 2023-03-19:
Thanks @GauravKumar9920 , I've merged #1965. I'm able to spawn the world file correctly.

GauravKumar9920, 2023-03-19:
Okkayy, I will look into it.

GauravKumar9920, 2023-03-19:
hey @pariaspe , can you provide me with the code to regenerate the bug.

pariaspe, 2023-03-19:
> hey @pariaspe , can you provide me with the code to regenerate the bug.

I'm building the RADI, launching the exercises and slowly moving the car (`HAL.setV(0.1)`).

GauravKumar9920, 2023-03-21:
Hey @pariaspe , I ran the bot for a few minutes and also restarted the same a few times the connection seems to be fine and  didn't get disconnected.

GauravKumar9920, 2023-03-24:
About the 

> Thanks @GauravKumar9920 , I've merged #1965. I'm able to spawn the world file correctly. I'm able to move the car too. However, connection closes after a couple of minutes of running the code. Also, JS map viewer is not working while the laser scan seems to be working on gazebo.

about js map viewer issue can you elaborate.
ps: I am done with exams, I can contribute actively.....

Amglega, 2023-03-26:
I changed the topic in the gui.py to "/catvehicle/front_laser_points" and the map shows the front laser at first but it does not get updated properly when the car starts moving:
[image]

GauravKumar9920, 2023-04-18:
I will try fixing it..

ango1994, 2023-06-22:
Closed on #2134

closed 2023-06-22 by ango1994