JdeRobot/RoboticsAcademy #1963: Unified RADI launching instructions

pariaspe, 2023-03-06:
Change launching instructions to an unique and simple command for all the exercises. Launching details should be moved to a common page to all.

AsRaNi1, 2023-03-06:
If it is alright may i work on this issue as well?
To lay out the idea though, the launching of RADI thought docker needs to be shifted to a different page and those instructions should be linked to all exercises?

pariaspe, 2023-03-06:
Yes, that's the main idea. Besides, we should find a common command to run all the "running" exercises (the ones in green in the grid).

AsRaNi1, 2023-03-07:
Yes, thank you for clearing it, I will soon submit a PR

pawanw17, 2023-03-14:
Hi, I am unable to find the pages where exercises are documented, Is it possible that [this guide](https://jderobot.github.io/RoboticsAcademy/contribute/) is outdated?

jmplaza, 2023-03-15:
AFAIK the instructions are scattered (and maybe outdated) along the [exercises web-pages](https://jderobot.github.io/RoboticsAcademy/exercises/). For instance the RADI launching instructions for FollowLine exercise can be found [here](https://jderobot.github.io/RoboticsAcademy/exercises/AutonomousCars/follow_line/#instructions); the instructions for the BasicVacuumCleaner exercise can be found [here](https://jderobot.github.io/RoboticsAcademy/exercises/MobileRobots/vacuum_cleaner#instructions); etc...

In addition, you may use some exercises at [Unibotics](https://unibotics.org/) and the updated RADI launching instructions appear in a modal before starting each exercise.

This issue is about gathering and updating all (most) of them and creating a single page with the RADI launching instructions.

pawanw17, 2023-03-15:
Hi @jmplaza , Actually I was talking of the source code where I could find that for the exercises documentation

jmplaza, 2023-03-15:
GitHub Pages branch (gh-pages) of RoboticsAcademy repository: https://github.com/JdeRobot/RoboticsAcademy/tree/gh-pages

closed 2023-03-19 by pariaspe