JdeRobot/RoboticsAcademy #1964: Minor frontend refinements for FollowLine+React+RAM exercise

jmplaza, 2023-03-08:
- [ ] Sim viewer activated by default and shown in the top part of the right side of the page
- [ ] RTF and measured BrainFreq in the second horizontal bar. No third horizontal bar at the footer anymore
- [ ] world selector in the first horizontal bar, aside the "Ready LED"
- [ ] "View Camera" label ----> GUI. So it can be used in all exercises
- [ ] "View Console" label ----> Console
- [ ] swap of "Theory" and "Code" buttons at the first horizontal bar
- [ ] Execution buttons: Play/Pause and Reset, both with icons instead of texts. The Play and Pause will be together in the same button, showing one or another icon depending on the 'state' of the execution

ango1994, 2023-03-13:
HI @yashphalle! I'm already working on this issue

yashphalle, 2023-03-13:
> HI @yashphalle! I'm already working on this issue

Okk no problem!
Let me know if there is something that I can do or help you!

closed 2023-03-16 by pariaspe