JdeRobot/RoboticsAcademy #1973: [Computer Vision exercises] Unable to launch Computer vision exercises that rely on webcam for input

pawanw17, 2023-03-15:
Hi, I am unable to launch Computer vision exercises that require user webcam (/dev/videoX) for input. I don't see the camera output on the web UI.

Command used to run:
`docker run --rm -it -p 8000:8000 -p 2303:2303 -p 1905:1905 -p 8765:8765 -p 6080:6080 -p 1108:1108 --device /dev/video0:/dev/video0 jderobot/robotics-academy`

I have tried with /dev/video1 and /dev/video0 with no luck.

[image]

codezerro, 2023-03-22:
i face the same issue

Qi-Zha0, 2023-03-27:
I hace the same issue. You could run  `v4l2-ctl --list-devices` and see a list of your video devices, make sure the camera you use is listed as /dev/video0

saurabhmj11, 2023-04-04:
It looks like you are running a Docker container for the Robotics Academy, and trying to use the webcam as an input to some Computer Vision exercises. However, you are not able to see the camera output on the web UI.

Here are some steps you can try to fix this issue:

Check if the webcam is working: Make sure that the webcam is properly connected and working. You can try running the webcam outside the Docker container to verify this.

Check if the webcam is being detected by Docker: Verify that the Docker container has access to the webcam by running the following command: ls -l /dev/video*. This should show the list of video devices available in the container. If the webcam is not listed, you might need to modify the Docker run command to include the correct device path.

Check if the exercise requires camera input: Verify that the Computer Vision exercise you are trying to run actually requires webcam input. Some exercises might work with pre-recorded video or image input instead of live camera input.

Check the web UI settings: Make sure that the web UI settings are configured to use the webcam as the input source. Check the settings of the exercise and make sure that the camera is selected as the input source.

Try running the container with elevated privileges: Some Docker containers require elevated privileges to access hardware devices such as webcams. You can try running the container with the --privileged flag to see if this resolves the issue.

Check for errors: Check the console or error logs for any errors or warnings related to the webcam. Fix any issues that might be preventing the webcam from working properly.

If none of the above steps work, try providing more details about the specific Computer Vision exercise you are trying to run, the webcam model and brand, and any error messages or warnings you receive. This will help in identifying the specific issue and providing a more targeted solution.

jmplaza, 2023-04-04:
That exercise heavily depends on the user machine's OS. It should work fine for Linux machines, but I have not seen it working on Windows machines due to problems in the access to USB ports of the host machine from the Robotics Academy Docker Image (RADI).

pawanw17, 2023-04-04:
Hi @jmplaza I am using ubuntu at the moment, I can give a shot to @saurabhmj11 's suggestions

pawanw17, 2023-04-06:
@jmplaza Still unable to view the user's camera in the UI. An interesting thing is, post the exercise launch the camera does get busy, i.e I am unable to use it on the host machine (the camera light is not on yet). After exiting the exercise, I am able to use it again. I am also able to run the camera in the RADI using a test script (with camera light on).

pawanw17, 2023-04-10:
@jmplaza The exercise seems to run fine in melodic RADI, Looking more into this.

jmplaza, 2025-03-20:
This is not required anymore as the ColorFilter exercise has been superseeded by the [Basic Computer Vision exercise](https://jderobot.github.io/RoboticsAcademy/exercises/ComputerVision/basic_computer_vision) (with real cameras :-) ).

It has been completely refactored using WebRTC technology to get the camera frames from the browser. This way it works not only for Linux but also for Windows and MacOS machines. The camera support using ROS drivers inside the RADI or RoboticsBackend docker container works for Linux, but not for Windows machines. It is due to security reasons preventing the mapping of USB ports from the docker container in Windows machines.

closed 2025-03-20 by jmplaza