JdeRobot/RoboticsAcademy #1975: Cleaning the repo

jmplaza, 2023-03-17:


RajGavadiya7, 2023-03-27:
Sir, I am participating in GSOC 2023, Can I work on this?

jmplaza, 2023-03-27:
Hi @RajGavadiya7 , 

there is no need to ask for permission to work on any issue, just submit your Pull-Requests. But in order to make sensible a Pull-Request for this issue long experience with the RoboticsAcademy is required I guess. This is not the best issue for newcomers. 

The best way to start contributing to RoboticsAcademy is to get familiar with this software product, just USE RoboticsAcademy, solve and provide feedback about the exercises already available in the current release (FollowLine, ObstacleAvoidance, VacuumCleaner, LocalizedVacuumCleaner...). They use the browser as the only GUI (for source code editing and for exercise monitoring), and a docker image for running the Gazebo simulator (it is named RADI= RoboticsAcademy Docker Image).

pariaspe, 2023-04-17:
First PR is ready to merge #1984. It cleans 30 out of 50 exercises.

## Exercises cleaned

- 3d_reconstruction
- autoparking
- autoparking_v2
- car_junction
- color_filter
- dl_digit_classifier
- drone_cat_mouse
- drone_gymkhana
- drone_hangar
- follow_line
- follow_line_newmanager
- follow_road
- follow_turtlebot
- global_navigation
- human_detection
- labyrinth_escape
- laser_loc
- laser_mapping
- montecarlo_visual_loc
- obstacle_avoidance
- opticalflow_teleop
- package_delivery
- position_control
- power_tower_inspection
- rescue_people
- road_junction
- vacuum_cleaner
- vacuum_cleaner_loc
- vacuum_cleaner_newmanager
- visual_lander

I've seen some small bugs in several exercises, I'll open a few issues as soon as we merge it. Also, I've seen some broken exercises:
- **laser_loc** and **laser_mapping**:
```
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/RoboticsAcademy/exercises/static/exercises/laser_loc/gui.py", line 184, in run
    self.gui.update()
  File "/RoboticsAcademy/exercises/static/exercises/laser_loc/gui.py", line 78, in update
    self.payload["pos_vertices"], self.payload["sonar_sensor"] = self.map.setSonarValues()
  File "/RoboticsAcademy/exercises/static/exercises/laser_loc/map.py", line 198, in setSonarValues
    hipotenusa = distance/(math.cos(cone/2))
TypeError: unsupported operand type(s) for /: 'list' and 'float'
```
- **digit_classifier**, **human_detection**, **color_filter** and **opticalflow_teleop**: Not able to load code or open camera device. I might be doing something wrong.

It will be great if you could test them before merging @dvalladaresv @ango1994 @ReyDoran . Thanks.
Apart from those exercises, I've done some testing and they should be working. Feel free to also test them.

-----

## Pending exercises
### *_react exercises will be next
- 3d_reconstruction_react
- autoparking_react
- dl_digit_classifier_react
- follow_line_react
- global_navigation_react
- obstacle_avoidance_react
- power_tower_inspection_react
- rescue_people_react
- vacuum_cleaner_react

### Prototype exercises
- 2d_visual_odometry
- amazon_warehouse (ros2 ¿?)
- bump_and_go
- follow_face
- machine_vision
- mobile_manipulation
- pick_place
- follow_line_game

Will it be worthy to clean them? There are exercises in other branches related to other ros distributions (mostly ros2 foxy). Again, should we cleaned them?

### Not listed in [exercises grid](https://jderobot.github.io/RoboticsAcademy/exercises/) neither included in the ddbb
- chrono (not migrated to web-templates. Is it Qualifying Formula1 exercise?)
- follow_line_turtlebot
- car_junction (already cleaned)

@jmplaza should we remove them?

### *_rotors exercises
- drone_cat_mouse_rotors
- follow_turtlebot_rotors

*_rotors exercises are deprecated since we are not using rotors anymore. Should we remove them ?

ReyDoran, 2023-04-18:
I have tested the migrated exercises. All of them seem to be working nicely besides the ones you mentioned. I've also found some minor bugs, I'm opening issues for them, but those were present before the migration.
Camera exercises aren't working for me either. Lastly, the exercise called Road Junction is using the id car_junction. The exercise with the id road_junction is the one missing in the db. Its template is using React but it seems it is still being developed.

pariaspe, 2023-05-15:
Exercises:
- [ ] 2d_visual_odometry
- [x] 3d_reconstruction
- [ ] 3d_reconstruction_react
- [ ] amazon_warehouse
- [x] autoparking
- [ ] autoparking_react
- [x] autoparking_v2
- [ ] bump_and_go
- [x] car_junction
- [ ] chrono
- [x] color_filter
- [x] dl_digit_classifier
- [ ] dl_digit_classifier_react
- [x] drone_cat_mouse
- [ ] drone_cat_mouse_rotors
- [x] drone_gymkhana
- [x] drone_hangar
- [ ] follow_face
- [x] follow_line
- [x] follow_line_newmanager
- [x] follow_line_react
- [ ] follow_line_turtlebot
- [x] follow_road
- [x] follow_turtlebot
- [ ] follow_turtlebot_rotors
- [x] global_navigation
- [ ] global_navigation_react
- [x] human_detection
- [x] labyrinth_escape
- [x] laser_loc
- [x] laser_mapping
- [ ] machine_vision
- [ ] mobile_manipulation
- [x] montecarlo_visual_loc
- [x] obstacle_avoidance
- [ ] obstacle_avoidance_react
- [x] opticalflow_teleop
- [x] package_delivery
- [ ] pick_place
- [x] position_control
- [x] power_tower_inspection
- [ ] power_tower_inspection_react
- [x] rescue_people
- [ ] rescue_people_react
- [x] road_junction
- [x] vacuum_cleaner
- [x] vacuum_cleaner_loc
- [x] vacuum_cleaner_newmanager
- [x] vacuum_cleaner_react
- [x] visual_lander

pariaspe, 2023-05-27:
## Remaining exercises
### Industrial Robots
- Pick Place
- Machine Vision
- Mobile Manipulation

Exercises are prototypes but already in web-templates. So no need to clean them.

### Service Robots
- Bump and Go
- Amazon Warehouse

Exercises are prototypes and not migrated to web-templates.

### Computer Vision
- 2D Visual Odometry
- Follow Face

Exercises are prototypes and not migrated to web-templates.

### Not Listed
- Chrono
- Follow Line Turtlebot

Not migrated to web-templates.

I'm closing this issue. Remaining exercises no need cleaning or are pending to migrate.

closed 2023-05-27 by pariaspe