JdeRobot/RoboticsAcademy #1986: Instructions to launch exercise with GPU is missing

iamarkaj, 2023-03-26:
Broken link: https://jderobot.github.io/gpu_acceleration

[image]

[image]

pawanw17, 2023-03-27:
Hi thanks for reporting, the correct link should have been https://jderobot.github.io/RoboticsAcademy/gpu_acceleration/

jmplaza, 2023-03-27:
Uhm, instead of having a specific webpage for this, perhaps we should add the GPU launching instructions to the ["Installation and running instructions" webpage](https://jderobot.github.io/RoboticsAcademy/installation/). Just to have all the information regarding using RoboticsAcademy in a single page....

Maybe we could rename that webpage as "User guide", don't you think?

pawanw17, 2023-03-27:
It is there as well in the end. as a link ... more on this here: https://github.com/JdeRobot/RoboticsAcademy/pull/1972#pullrequestreview-1345535208

pawanw17, 2023-03-27:
A quick fix to this issue would be to change /gpu_acceleration to /RoboticsAcademy/gpu_acceleration in the pages where it is used eg: https://raw.githubusercontent.com/JdeRobot/RoboticsAcademy/gh-pages/_pages/installation.md here however that'd make it not run in the development env.

pawanw17, 2023-03-28:
@pariaspe @jmplaza  Do you think I should remove these lines from the exercises?

```
Enable GPU Acceleration
Follow the advanced launching instructions from [here](gpu_acceleration).
```
or replace the link with /user_guide... If I do replace it then there'll be the same problem facing currently

https://github.com/JdeRobot/RoboticsAcademy/blob/gh-pages/_pages/exercises/AutonomousCars/autoparking.md#enable-gpu-acceleration

jmplaza, 2023-03-28:
Yes, in the exercise webpages a link to /user_guide page should be included, which has to be a permalink (replacing /installation).

In the user_guide the instructions with the URL to use from the browser to start using RoboticsAcademy should be clarified too.

I guess we don't need a different environment for users and for developers of RoboticsAcademy. The developer may modify the source code of the webserver or a given exercise, and launch the webserver at the same port. Anyway we could include a /developer_guide webpage with the instructions to deploy RoboticsAcademy without creating a RADI, just launching django locally....

pawanw17, 2023-03-28:
> Yes, in the exercise webpages a link to /user_guide page should be included, which has to be a permalink (replacing /installation).
> 
> In the user_guide the instructions with the URL to use from the browser to start using RoboticsAcademy should be clarified too.
> 
> I guess we don't need a different environment for users and for developers of RoboticsAcademy. The developer may modify the source code of the webserver or a given exercise, and launch the webserver at the same port. Anyway we could include a /developer_guide webpage with the instructions to deploy RoboticsAcademy without creating a RADI, just copying the repositories and launching django locally....

Sorry I didn't mention clearly, if I add link /user_guide in the exercises page, the Website will work in development environment 127.0.0.1:4000/user_guide

however in the production it'll redirect to https://jderobot.github.io/user_guide, which will give page does not exist error,

If I instead add the link /RoboticsAcademy/user_guide in the exercises pages, then it'll work in the production environment but not in the development environment since 127.0.0.1:4000/RoboticsAcademy/ does not exist.... This is the issue currently facing with the gpu_acceleration page

pawanw17, 2023-03-28:
Um to edit exercises documentation I go to gh-pages branch and run
`bundle exec jekyll serve`

Upon doing so on 127.0.0.1:4000 this is the website django server not the RADI one

[image]
[image]

To edit exercise documentation I have used this http://jderobot.github.io/RoboticsAcademy/contribute/#jekyll-local-installation

I am not running the RADI, or the RADI django server which runs on 8000

jmplaza, 2023-03-28:
@pawanw17 check my previous updated comment. 

There was a confusion in our conversation, please notice there are two sides here: 
(a) DOCUMENTATION 
the update of GitHub Pages documentation  (calling the local deployment of GitHub Pages with Jekyll templates a "development environment" is too ambitious...) 

(b) OPERATING ROBOTICSACADEMY 
the development environment for RoboticsAcademy developers (refining of exercises, of the django webserver, creation of new exercises).

Hope it is more clear now.

pawanw17, 2023-03-28:
yes understood, the cause of confusion was "development environment" XD, I'll make a PR soon, Thanks for clearing things out!

jmplaza, 2023-03-28:
https://forum.unibotics.org/ is up and running AFAIK

pawanw17, 2023-03-28:
yeah but the website redirects to https://forum.jderobot.org/c/english/roboticsacademy

jmplaza, 2023-03-28:
please just correct the link to https://forum.unibotics.org/ in the PR

jmplaza, 2023-03-28:
Fixed in #1988 and #1989. Well done @pawanw17 !